TheThingsNetwork/lorawan-stack #5465: Show end device status information as part of the end device overview

adriansmares, 2022-05-18:
#### Summary

We should show device status information as part of the end device overview in the Console.

#### Why do we need this?

The background here is that we have a mechanism that exists at network layer (so between the Network Server and the end device) to find out a bit of information about the state of the end device. This information is periodically retrieved by the Network Server and stored in the following fields in the NS end device registry:
- `power_state` - an enum that basically allows the end device to signal where it is drawing power from (battery, wall socket, unknown)
- `battery_percentage` - battery percentage when a battery is used
- `downlink_margin` - signal to noise ratio, in dBm, of the last downlink received by the end device
- (optional) `last_dev_status_received_at` - the timestamp at which a device status response was received
- (optional) `last_dev_status_f_cnt_up` - the frame counter at which a device status response was received

#### What is already there? What do you see now?

The end device overview shows activation information, session information etc.

#### What is missing? What do you want to see?

Have the device status information available, maybe under the map ?

#### Environment

`v3.19.2`, but not relevant

#### How do you propose to implement this?

'Just show the fields.'

All UX jokes aside, I don't have more ideas other than having a device status paragraph in the overview.

#### How do you propose to test this?

Most of the end devices on `staging1` do have these fields filled in, so you can check the output there.

#### Can you do this yourself and submit a Pull Request?

Can review.

kschiffer, 2022-05-18:
Great idea. I'll wireframe something and schedule this.

NicolasMrad, 2023-04-12:
Since now we have the NOC, can't we display the above there?

adriansmares, 2023-04-12:
> Since now we have the NOC, can't we display the above there?

The information is complementary - the Console can show the latest information, while the NOC may show the historical evolution of said information.

KrishnaIyer, 2023-07-04:
This is a good one indeed. Let's try to get the wireframes for Q3.

kschiffer, 2023-08-31:
I've finished the design of this as part of the general Console UX update. Implementatino planning follows.

KrishnaIyer, 2024-10-10:
This is done as part of the redesign

closed 2024-10-10 by KrishnaIyer