TheThingsNetwork/lorawan-stack #5491: Report gateway IP address in gs.gateway.connect event

furtiman, 2022-05-26:
#### Summary

Add gateway IP address to `gs.gateway.connect` event.
Can be reliably determined for UDP, Basic Station and gRPC gateways but not for MQTT gateways.

#### Why do we need this?

To be able to use / display IP of the gateway to users.

#### What is already there? What do you see now?

`evtGatewayConnect` not containing IP address

#### What is missing? What do you want to see?

IP address added to payload of the event.

#### Environment

3.19.2

#### How do you propose to implement this?

Add extra payload and change event data type in `gs.gateway.connect` event (`pkg/gatewayserver/observability`)

#### How do you propose to test this?

UT?

#### Can you do this yourself and submit a Pull Request?

re @johanstokking will put you for now

htdvisser, 2022-05-27:
I think we can try adding the gateway IP address to the `GatewayConnectionStats` message.

The only question is if/how this is going to work when The Things Stack is behind a load balancer / proxy. Then we can't reliably determine the original IP address (unless we add PROXY protocol support).

KrishnaIyer, 2022-05-30:
> The only question is if/how this is going to work when The Things Stack is behind a load balancer / proxy. Then we can't reliably determine the original IP address (unless we add PROXY protocol support).

Indeed and additionally, if gateways are using a VPN then this won't work either. This is also one of the reasons we decided not to use IP addresses for location data.

I believe that this should be a sub-component of the broader gateway management issue; https://github.com/TheThingsIndustries/lorawan-stack/issues/2694

johanstokking, 2022-06-04:
> @htdvisser  I think we can try adding the gateway IP address to the `GatewayConnectionStats` message.

Yes, for it to be requested from GS, for the Console. We don't have to put the IP address in every stats update (to store or events); only on connect. We could add the message as event payload to `gs.gateway.connect` (currently there's no payload), and include the IP address there.

> @htdvisser  The only question is if/how this is going to work when The Things Stack is behind a load balancer / proxy. Then we can't reliably determine the original IP address (unless we add PROXY protocol support).

It becomes a bit deployment specific, but in case of UDP we already have the source IP address (even if AWS NLB is in between). In case of Basic Station and MQTT, indeed we'd need Proxy Protocol support. But I would argue that having the source IP address is useful anyhow.

> @KrishnaIyer I believe that this should be a sub-component of the broader gateway management issue; [TheThingsIndustries/lorawan-stack#2694](https://github.com/TheThingsIndustries/lorawan-stack/issues/2694)

How so? The use case here is to show the gateway's IP address in the NOC and potentially Console.

htdvisser, 2022-06-07:
> It becomes a bit deployment specific, but in case of UDP we already have the source IP address (even if AWS NLB is in between). In case of Basic Station and MQTT, indeed we'd need Proxy Protocol support.

Indeed, in our AWS ECS deployments we should be able to determine the source IP address, also for MQTT, which uses pretty much the same AWS NLB settings as UDP. For Basic Station, Envoy should also get the correct source IP from AWS NLB, and add an X-Forwarded-For header that we can read in The Things Stack.

The problem is mostly with other (unknown) proxy configurations, but I suppose we can just get away with saying that this feature only works reliably in officially supported deployment models.

johanstokking, 2022-06-07:
> The problem is mostly with other (unknown) proxy configurations, but I suppose we can just get away with saying that this feature only works reliably in officially supported deployment models.

Yep. Including not using a load balancer, which isn't really needed with TTSOS anyway.

So we the origin IP already for UDP, Basic Station, GRPC and MQTT. The only thing is that the frontends need to provide the IP address nicely, probably with a new context key.

johanstokking, 2022-06-07:
@KrishnaIyer please assign a version milestone or we discuss in next weekly call.

KrishnaIyer, 2022-06-08:
I've added `v3.20.2` taking holidays into account.

KrishnaIyer, 2022-07-05:
Proposal looks good to me. However, I think that adding the new struct (or `net.Addr`) as an argument to `Connect` seems better to me instead of passing it in the context.

PR #5581 (2022-07-12): Add Gateway IP address to gateway connection stats

#### Summary

Closes #5491 

#### Changes

- Add the Gateway Remote Address message to `GatewayConnectionStats`
- Implement changes in GS
- Update frontends to report the IP address.
- Tests

#### Testing

- [x] Local Testing

Stats
```
Before Connection
~ » ttn-lw-cli gtw get-connection-stats test-gtw
WARN	Using insecure connection to OAuth server
WARN	Using insecure connection to API
error:pkg/gatewayserver:not_connected (gateway `test-gtw` not connected)
    gateway_uid=test-gtw
    correlation_id=62fdc3e979fb4c5989e4d693d18c00d0
--- error [... cut]
closed 2022-07-12 by KrishnaIyer