TheThingsNetwork/lorawan-stack #5501: Allow multi-domain usage for the console

philippdzm, 2022-05-30:
#### Summary

Allow multi-domain configuration for TheThingsStack console.

#### Why do we need this?

For deployments in intranets it may be relevant to access the console via an intranet domain and a domain which works via VPN. Other use cases may exist as well.

#### What is already there? What do you see now?

N/A

#### What is missing? What do you want to see?

Being able to open and use the console via different domains -- means also the relevant web APIs should be accessible.

#### How do you propose to implement this?

It is difficult to propose a solution as the hole cluster oriented architecture appears incompatible to the requested feature. 
But maybe it is possible to allow multiple URLs for the web UI.
https://github.com/TheThingsNetwork/lorawan-stack/blob/v3.20/config/stack/ttn-lw-stack-docker.yml#L80

#### How do you propose to test this?

Create a deployment with multiple domains and verify that using the console by accessing through both domains works.

#### Can you do this yourself and submit a Pull Request?

No, not sure how to approach this as not completely understanding the design of the architecture.

NicolasMrad, 2022-05-31:
thank you for submitting this feature request.
at the moment this feature is not in our pipeline but we will add it to our backlog and schedule it in the future

KrishnaIyer, 2025-04-30:
This isn't something we are likely to support.