TheThingsNetwork/lorawan-stack #5512: Load device profile from DR when using CSV import

kschiffer, 2022-06-02:
#### Summary
When importing end devices in bulk using the end device import feature, the device profile should be applied as provided from the DR.

#### Why do we need this?

- Facilitate device import, allow importing devices from keys only, without the need of providing settings manually
- Make better use of data we have in the DR

#### What is already there? What do you see now?

- Device profiles in the DR
- End device import feature
- The device selector from the single device onboarding

#### What is missing? What do you want to see?

- Automatically apply device profile (mac settings, MAC/PHY, FP, etc) to the imported devices
- Add the device model selector from the single device onboarding to the device import UI (so it can be used as fallback)

#### How do you propose to implement this?

I see two ways:

1. Load the profile in the frontend and apply it before running the creation API requests
2. Load the profile in the backend (EDTC) and apply it to the output

I believe that is more desirable because it would be applicable for CLI as well. We might also want to allow disabling this automatic device profile application if wished.

#### Can you do this yourself and submit a Pull Request?

I can do the frontend part but I need help for adding the backend functionality.

cc @adriansmares

nicholaspcr, 2022-07-11:
When working on this I had an error when making a request to the Device Repository in which it states that `rights` are needed for the request.
```
// FetchProfile provides the end device profile.
func (dtc *DeviceTemplateConverter) FetchProfile(device *ttnpb.EndDevice) (*ttnpb.EndDeviceTemplate, error) {
	conn, err := grpc.Dial(dtc., grpc.WithInsecure())
	if err != nil {
		logger.Error("NICK -> ", err)
		return nil, err
	}

	return ttnpb.NewDeviceRepositoryClient(conn).GetTemplate(context.Background(), &ttnpb.GetTemplateRequest{
		ApplicationIds: device.GetIds().GetApplicationIds(),
		VersionIds:     device.GetVersionIds(),
	})
}
// Generates
ERROR  rpc error: code = PermissionDenied desc = error:pkg/auth/rights:no_application_rights (no rights for application `test-app`)
```
After having a talk with @adriansmares the request of application rights on the DR seems to be a bit strange, when the device repository was specified by https://github.com/TheThingsNetwork/lorawan-stack/issues/3433, one of the comments were `The API in principle doesn't require rights, but do require any authentication`.

Two are two options to change this are:
1. Use the [rpcnetadata.WithForwardedAuth](https://github.com/TheThingsNetwork/lorawan-stack/blob/a108016b36b5253c05673a75c6e2f03c038bc0f0/pkg/rpcmetadata/request.go#L39-L47) to validate the authenticate the request.
2.  Deprecate the requirement of having the application identifiers on DR requests. 
 
@johanstokking, wanted to get your opinion on the two options.

johanstokking, 2022-07-12:
We want some form of authentication for rate limiting user tokens or API keys. Also external services would be able to use TTSC as a public API for querying the device repository. So we must keep authentication. You can document the reasons for this so our future selves know why we do this. I very much understand the question.

@KrishnaIyer didn't you need a simple "is authenticated" check too? Did you end up writing something reusable? Don't we simply want something in `rights.RequireAuthenticated()`, which may indeed get auth info or something similar to what forwarded auth attempts.

KrishnaIyer, 2022-07-13:
[rpcmetadata.WithForwardedAuth](https://github.com/TheThingsNetwork/lorawan-stack/blob/a108016b36b5253c05673a75c6e2f03c038bc0f0/pkg/rpcmetadata/request.go#L39-L47) already does this. It checks the incoming context for "any" auth which includes ClusterAuth.

The reason to check for Application ID rights was that the call to `GetTemplate` is designed to be used in the device onboarding flow where the client (CLI/Console) has both the user auth and the Application ID under which the device would be created.

Now if a cluster component also calls DTC to fetch the device template, I would prefer adding an additional check for ClusterAuth (like in [`GetUplinkDecoder`](https://github.com/TheThingsNetwork/lorawan-stack/blob/a108016b36b5253c05673a75c6e2f03c038bc0f0/pkg/devicerepository/grpc.go#L176-L180)) instead of "any" auth.

johanstokking, 2022-07-13:
> [rpcmetadata.WithForwardedAuth](https://github.com/TheThingsNetwork/lorawan-stack/blob/a108016b36b5253c05673a75c6e2f03c038bc0f0/pkg/rpcmetadata/request.go#L39-L47) already does this. It checks the incoming context for "any" auth which includes ClusterAuth.

Yes, but that is not a idiomatic `rights.RequireXXX()` check.

> The reason to check for Application ID rights was that the call to `GetTemplate` is designed to be used in the device onboarding flow where the client (CLI/Console) has both the user auth and the Application ID under which the device would be created.

All the other rpcs need auth too, and that is not for application context, but because we don't want to allow anonymous access to this API.

KrishnaIyer, 2022-07-13:
Ok so we'd need a `rigths.RequireAny()` or something that wraps [rpcmetadata.WithForwardedAuth](https://github.com/TheThingsNetwork/lorawan-stack/blob/a108016b36b5253c05673a75c6e2f03c038bc0f0/pkg/rpcmetadata/request.go#L39-L47) then?

johanstokking, 2022-07-13:
Why wrap getting a call option?

We just need:

```go
	md := FromIncomingContext(ctx)
	if md.AuthType == "" || md.AuthValue == "" {
		return errUnauthenticated.New()
	}
	return nil
```

adriansmares, 2022-07-13:
> Why wrap getting a call option?
> 
> We just need:
> 
> ```go
> 	md := rpcmetadata.FromIncomingContext(ctx)
> 	if md.AuthType == "" || md.AuthValue == "" {
> 		return errUnauthenticated.New()
> 	}
> 	return nil
> ```

What is the point of requiring that values exist if they may be `foo` and `bar` ? Just for logging and tracing ? 

For rate limitting concerns if the auth value is randomized, and not validated, the rate limitting is basically neutralized.

johanstokking, 2022-07-14:
Right, so we need to actually get the auth info.

nicholaspcr, 2022-07-18:
I meant to write here earlier but opted to work on the PR draft to get a better idea of the whole flow and see if I can get the entire picture. In short, there is the validation to see if the authentication exists and its there to impede other people from using or spam the API.

>The reason to check for Application ID rights was that the call to GetTemplate is designed to be used in the device onboarding flow where the client (CLI/Console) has both the user auth and the Application ID under which the device would be created.

While this is true for the DTC as well, considering that console/api would have access to both information, I'm not sure why demand the application-ID in this request. Shouldn't the authentication validation be enough to confirm that the request is being done by an authenticated user? This feels a bit out of the context of the DR/GetTemplate and forces any other service that would want to fetch a profile to have the application-ID as well, meaning that in this case the `application-id` serves as an authentication procedure.

So ideally there should be an auth validation in the `WithForwardedAuth` that would go beyond the check of `if fields aren't empty`, with this the necessity of having authentication would be good to go. I might have misunderstood how the validation process goes so if anything that I typed seems out of place let me know.

<hr/>

@kschiffer  you mentioned in the description of having something of a selector to serve as fallback when importing devices, on the PR I initially took the approach of the user having the responsibility of inserting the `vendor_id` and `vendor_profile_id` into the csv contents and that way he could specify models to each of the devices. That also allowed me to **not** include the such fields on the request message and keep API changes to a minimal.

Wanted to get your opinion if this is good UX wise or if there should be both options, meaning that there would be a dropdown where the user can select the profile template in which it will be used as a fallback but he could also setup which profile to fetch via the csv content.

johanstokking, 2022-07-19:
> While this is true for the DTC as well, considering that console/api would have access to both information, I'm not sure why demand the application-ID in this request. Shouldn't the authentication [...]

Fully agreed.

> So ideally there should be an auth validation in the `WithForwardedAuth` that would go beyond the check of `if fields aren't empty`, with this the necessity of having authentication would be good to [...]

Yes, we should get the auth info with `rights.AuthInfo()`. It returns a non-nil struct (and nil error) if there's authentication. If the request is unauthenticated, it returns a nil struct. If there's invalid auth, it returns an error. See https://github.com/TheThingsNetwork/lorawan-stack/blob/v3.21/pkg/auth/rights/auth_info.go#L41-L48.

> It would be a convenient thing but not strictly required. I could envision that some users only have a plain list with only the provisioning data. If they know the device type of the fleet, they could simply add it in the UI instead of having to manually add it to the CSV.

I think it's a common use case to onboard a bunch of the same devices. So selectors for indicating the device that work as a fallback for what is in the file sounds like a great idea.

nicholaspcr, 2022-07-29:
I updated the PRs related to this issue.

The deprecation of the ApplicationID on the Device Repository is handled #5659 and the actual loading of the end device template is handled by #5610.

Like I wrote in #5610 I ended up separating the changes into different PRs, from the discussion we had the changes related to this functionality were:
- The functionality itself (loading end device template)
- Deprecating the ApplicationID from the DR.
- Updating the `rpcmetadata.WithForwardedAuth`

The last topic currently there isn't a PR for it yet, the changes would be:
- The `rpcmetadata` is used by the `rights` pkg, causing a dependency cycle. This can be solved by just by creating the method on the `rights` pkg and replacing the `rpcmetadata.WithForwardAuth` usage with the new `rights` function.
- The changed of checking for rights instead of if the metadata isn't empty implies in updating the unit tests in a similar fashion to how it was done #5659. Which requires updates to the unit tests of `qrcodegeneartor`, `dcs` and `gs`, since they use `rpcmetadata.WithForwardAuth`.

Since the change of the `WithForwardedAuth` isn't something that would block this issue I'm incline to just write it as a  `technical_debt` issue.

cc: @johanstokking

johanstokking, 2022-07-31:
I think we should use `rpcmetadata.WithForwardedAuth` for forwarding auth when there are call being made from an already authenticated context (so in a service implementation calling other services).

The rights require check should be, like the others, in `pkg/rights`. This should forward the auth to get the `AuthInfo`, and check there response or error code.

nicholaspcr, 2022-07-31:
I agree, these responsibilities for each package makes sense. 

I was trying to mention that the `auth/rights` pkg uses the `rpcmetadata` to validate if there is auth information before requesting the rights to the IS's `ApplicationAccessServer`. If we are changing the `rpcmetadata.WithForwardedAuth` to validate if the request has any rights, trying to call the `rights.AuthInfo` in the method would cause a dependency cycle. That's why I suggested having a `WithForwardedAuth` function on the `rights` pkg. Since that goes beyond the responsibility of the pkg itself, the other option would be then to remove the `rpcmetadata` usage from the `auth/rights` pkg.

I may be conflicting things but now that I think about it, it doesn't feel right to fetch which rights the user has in order to validate if he is authenticated or not. Feels like a mixture of Authentication and Authorization, we would be effectively verifying what the user is authorized to do in order to affirm that he is authenticated. In practice the validation that we want is just to know if he exists or not, fetching its rights feel a bit excessive.

johanstokking, 2022-08-16:
> If we are changing the `rpcmetadata.WithForwardedAuth` to validate if the request has any rights

We're not changing this. That method is perfectly fine; it's really about forwarding auth from an incoming context to an outgoing context and that's all that's relevant for the package.

> Feels like a mixture of Authentication and Authorization, we would be effectively verifying what the user is authorized to do in order to affirm that he is authenticated. In practice the validation that we want is just to know if he exists or not, fetching its rights feel a bit excessive.

Right. The thing is that if the client passes any token on the `authorization` header, the service has to check whether the client is _authenticated_ still. The service needs to check the validity of the token. The only party that can and should do that, is Identity Server, and using `AuthInfo`. That may return more than you need, but most of all, if the token isn't valid at all, the service will know that. Indeed, authorization comes after, but that isn't really relevant here. For completeness, indeed we're talking about `rights.RequireAuthenticated()` (authentication) while the rights requirements are authorization.

nicholaspcr, 2022-08-25:
With #5610 merged, there is now support for loading the end-device profile from the csv fields or through a fallback value given in the [API](https://github.com/TheThingsNetwork/lorawan-stack/blob/0b60aec8c674cecc9125bebf34c7176580b90146/api/end_device.proto#L905)

I believe  the only section missing is:
>Add the device model selector from the single device onboarding to the device import UI (so it can be used as fallback)

@kschiffer I'll leave this to you or to whoever you assign it to, if there is anything not clear please let me know.

closed 2023-03-30 by ryaplots