TheThingsNetwork/lorawan-stack #5538: Uplinks of 255 bytes are discarded in EU868 and IN865 frequency plan

Achichig, 2022-06-15:
#### Summary

Device configured on LoRaWan v1.0.4 - RP2 1.0.1 on EU868 and IN865 frequency plan and ADR configured for using DR5. 
Uplink of 255 bytes (Phy_payload) are discarded when it shouldn't.

#### Steps to Reproduce

1. Configure device to DR5
2. Send an uplink with max payload size

#### What do you see now?

On stack logs I see:
`stack_1      | DEBUG    Uplink length exceeds maximum   {"ack": false, "adr": false, "adr_ack_req": false, "band_id": "IN_865_867", "bandwidth": 125000, "class_b": false, "data_rate": "lora:<bandwidth:125000 spreading_factor:7 > ", "dev_addr": "019F5B1E", "f_cnt_reset": false, "f_opts_len": 0, "f_port": 1, "frequency": 866585000, "full_f_cnt_up": 11, "grpc.method": "HandleUplink", "grpc.service": "ttn.lorawan.v3.GsNs", "last_f_cnt_up": 9, "m_type": "CONFIRMED_UP", "mac_version": "MAC_V1_0_4", "major": "LORAWAN_R1", "namespace": "networkserver", "pending_session": false, "phy_payload_len": 255, "received_at": "2022-06-15T09:33:53.976233982Z", "request_id": "01G5KBCMXRSCWK7J71QWHBYM4V", "spreading_factor": 7, "uplink_f_cnt": 11}`

According to the lora specification 1.0.4:
[image]

And the regional parameters:
[image]

M = 250 bytes on DR5
So Len(PHY_PAYLOAD) max = M + len(MHDR) + len(MIC) = 250 + 1 + 4 = 255 bytes

#### Environment

TTN v3.19.1

#### How do you propose to test this?

...

#### Can you do this yourself and submit a Pull Request?

Unfortunately not

adriansmares, 2022-06-21:
This is a valid bug report, caused by the way our stack handles payload limits.

Right now the payload limit for the IN865 band is based on the repeater-compatible scenario:
[image]
https://github.com/TheThingsNetwork/lorawan-stack/blob/ad93aca6244217d356231af47cb871fb43b9fa9f/pkg/band/in_865_867_rp2_v1_0_1.go#L40-L49
(In case anyone wonders why data rate 6 is missing - it is marked as RFU in the spec but they still provide the payload limits)

I think that maybe we should have a `MACSettings` field that enables this repeater limitation on/off, and to have both limits available as part of the band package. 

@johanstokking any thoughts on this ?

johanstokking, 2022-06-21:
I think we can assume that repeaters aren't being used for now.

When device-to-device specifications land in LoRaWAN or in a companion specification, we can revisit this.

closed 2022-07-01 by adriansmares