TheThingsNetwork/lorawan-stack #5542: Field masks for event streams

kschiffer, 2022-06-16:
#### Summary
We should provide a way to pass field masks to the event stream containing the desired fields for inspection, so that any unwanted data is stripped from the stream.

#### Why do we need this?
- Make it easier to retrieve relevant information
- Decrease bandwidth for streams
- To configure a default set of field masks and only show all data in a verbose mode

#### What is already there? What do you see now?
- Field masks for regular API
- Event streams and filtering out event types

#### What is missing? What do you want to see?
Field masks for event props

#### How do you propose to implement this?
I suppose the biggest challenge here is the ubiquity of different event props, which varies between event types. We'll likely want to make this in a freeform way, meaning any field mask value can be provided and the stream will use it as a filter for each event prop. To make this sufficiently useful, we would also need to make this work for lower-level props, e.g. `data.uplink_message.rx_metadata.snr`.

I cannot specify how exactly this should be implemented since I'm not working on event streams. I'll add a discussion labe until we worked out the exact location.

#### Can you do this yourself and submit a Pull Request?
No, but I'm happy to help testing. Assigning @johanstokking for now, who I believe came up with this idea. Please reassign as appropriate.

johanstokking, 2022-06-17:
Right, since there are different event types with each having their own event data types, we would need a map of event type to field mask as part of the subscription.

Did I come up with this? The only real benefit I see indeed is to reduce bandwidth and to harmonize this with getting entities (where the server also applies the field mask). Otherwise, I think that selecting which fields are relevant and removing noise is the job of the frontend.

Indeed, let's discuss this offline first.

kschiffer, 2022-06-17:
Actually, I think we should broaden the scope of this a bit. This is not only relevant for the event streams, but also for the event data passed to integrations such as webhooks, where users will be mostly interested in only some props but currently get a huge data object for each event. They can of course disregard unused data but it's still a lot of wasted bandwidth.

As far as the event streams in the Console go, this can indeed be solved on the frontend.

kschiffer, 2022-07-14:
Replaced by https://github.com/TheThingsNetwork/lorawan-stack/issues/5597

closed 2022-07-14 by kschiffer