TheThingsNetwork/lorawan-stack #5553: Potential regression in UpdateEndDeviceRequest

lulf, 2022-06-22:
#### Summary

We have a component that is reconciling the internal state of our device database with TTN, and it recently started failing with:

```

#### Steps to Reproduce
<!-- How can we reproduce the problem? -->

1. ...
2. ...
3. ...

<!--
Please upload relevant configuration (as .txt).
If you use the command "ttn-lw-stack config", you can redact sensitive config.
-->

#### What do you see now?
<!-- Please paste terminal output, upload logs (as .txt) or upload screenshots. -->

...

#### What do you want to see instead?
<!-- Please add some examples or mock-ups if applicable. -->

...

#### Environment
<!--
Your environment: OS/Browser/Gateway/Device/...? Versions? IDs/EUIs?
Paste the output of "ttn-lw-cli version" or "ttn-lw-stack version" if applicable.
-->

...

#### How do you propose to implement this?
<!-- Please think about how this could be fixed. -->

...

#### How do you propose to test this?
<!-- Please think about how this is verified as fixed. -->

...

#### Can you do this yourself and submit a Pull Request?
<!-- You can also @mention experts if you need help with this. -->

...

lulf, 2022-06-22:
Sorry for the spam. I accidentally hit create so I had to update the issue a few times

nejraselimovic, 2022-06-29:
@lulf Ulf, thank you for creating this issue. The `ids.dev_eui` field is indeed a part of the allowed field mask for the `Set` method, because you have to use it when registering the end device. However, when the end device is already created, this field can no longer be changed, i.e. you cannot update the DevEUI. This goes for app ID, device ID, join/app EUI, device EUI and API key ID. You can also notice that TTS Console prevents you from updating these fields as well. 

To close this issue, I'll make a PR with a note that IDs cannot be updated.

lulf, 2022-06-29:
@nejraselimovic Thanks for clarifying this! We'll update our usage of the API accordingly.

closed 2022-07-01 by nejraselimovic