TheThingsNetwork/lorawan-stack #5561: Re-enable Goreleaser parallelism during release

adriansmares, 2022-06-29:
#### Summary

References https://github.com/canonical/craft-cli/issues/84

#### Why do we need this?

The Goreleaser parallelism has been disabled since `snapcraft` does not support it due to the above mentioned issue. Once that issue is solved, we should re-enable parallelism.

#### What is already there? What do you see now?

The `goreleaser` step using the `release-tag.yml` action uses `--parallelism 1`. 

#### What is missing? What do you want to see?

The parallelism should not be set at all (this causes it to default to the number of available cores).

#### Environment

v3.20.1

#### How do you propose to implement this?

Remove the tag.

#### How do you propose to test this?

On the next release.

#### Can you do this yourself and submit a Pull Request?

Yes.

adriansmares, 2022-09-01:
@ysmilda please check if the latest release of `snapcraft` contains the upgraded [`craft-cli`](https://github.com/canonical/craft-cli/issues/84) such that we can remove this flag.

ysmilda, 2022-09-01:
The fixed CLI has been included in `snapcraft` since version `7.1.0` (currently on `7.1.3`). We should be able to remove the flag without problems.

PR #5754 (2022-09-01): Re-enable Goreleaser parallelism during release

#### Summary

Closes #5561 

#### Changes

- Re-enabled parallelism on Goreleaser

#### Testing

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Compatibility: The changes are backwards compatible with existing API, storage, configuration and CLI, according to the compatibility commitments in `README.md` for the chosen target branch.
- [x] Documentation: Relevant documentation is added or updated.
- [ ] Changelog: Significant features, behavior changes, deprecations and fixes are added to `CH [... cut]