TheThingsNetwork/lorawan-stack #5577: Downlink transmission failed with result Too EARLY

zouOrange, 2022-07-04:
### Summary

Hello, please ask a question, in Class C mode, TTN(NS) sends data too early(NS displays an error message: Failed to transmit downlink Downlink transmission failed with result Too EARLY), resulting in packet loss problems. Why? Thank you!

### Current Situation

Frame counter width
●32 bit
Desired Rx1 delay②
5  sec
Desired Rx1 data rate offset
0
Desired Rx2 data rate index
8
Desired Rx2 frequency
923.3MHz
Desired maximum duty cycle
100%
Factory preset frequencies
Class C timeout②
300 seconds
Status time periodicity
200
messages
86400 seconds

### Desired Situation

Want to give problem analysis and solutions.

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

KrishnaIyer, 2022-07-04:
@zouOrange: Please add proper debugging information that includes device, gateway settings and event logs.

zouOrange, 2022-07-04:
You mean I set the device information with an error?

zouOrange, 2022-07-04:
[image]
[image]
[image]
[image]
This is my configuration information, please take a look.

zouOrange, 2022-07-04:
This is where I went wrong
[image]

KrishnaIyer, 2022-07-06:
Are you using Class-C absolute scheduling? What is the scheduling request?

The Things Stack can buffer messages to be sent to gateways if the gateway rejects downlinks sent in advance.

So, on the console, you could select the "Enable server-side buffer of downlink messages" option and try again. 

Also, gateway Packet Forwarder logs would be really helpful.

KrishnaIyer, 2025-04-30:
This issue is stale.

closed 2025-04-30 by KrishnaIyer