TheThingsNetwork/lorawan-stack #5578: Loss of inventory in upgrade from 3.6 to 3.20

andreusoler8, 2022-07-04:
### Summary

I am following the official upgrade documentation (https://www.thethingsindustries.com/docs/getting-started/upgrading/) to upgrade from my current version of TTS 3.6.0 to the latest version available. Following the steps, I manage to update the version but neither the registered users, nor the gateways, nor the applications with their devices are maintained.
What should I do to be able to keep all my data without being deleted?

### Current Situation

Upgrade done successfully.

### Desired Situation

Update performed keeping all previous data.

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

NicolasMrad, 2022-07-05:
Thank you for submitting the issue @andreusoler8 , can you please provide logs for the upgrade process because it is not possible not upgrade to v3.30 directly from v3.6, you have to upgrade to 3.19 first.

adriansmares, 2022-07-19:
Hey @andreusoler8,

Could you please provide the logs for the actual migration operations ? (`is-db migrate`, `ns-db migrate` etc.) 

We cannot really tell what caused your inventory to be lost only from the Redis logs - they do not show any data loss themselves. Also be sure to change the image for Redis from `redis:latest` to `redis:6.2`.

danijelk, 2022-10-21:
@NicolasMrad 
> Thank you for submitting the issue @andreusoler8 , can you please provide logs for the upgrade process because it is not possible not upgrade to v3.30 directly from v3.6, you have to upgrade to 3.19 first.

Is the 3.19 jump documented anywhere, just upgraded to 3.22.1 but had issues as I had to do 3.19 migrations first and then jump to 3.22, without your comment I would probably have never found out I feel.

closed 2025-04-30 by KrishnaIyer