TheThingsNetwork/lorawan-stack #5585: Clarify the semantics of the uplink message timestamps

adriansmares, 2022-07-08:
### Summary

The timestamps found in the `UplinkMessage` message should have their semantics clarified.

### Current Situation

We currently have multiple timestamps available:

1. The timestamp at which the Gateway Server / Network Server received the frame:
https://github.com/TheThingsNetwork/lorawan-stack/blob/70d88dd86a9b17bf8faf1bbea6961d6c5d3f2ae7/api/messages.proto#L83-L85

2. The timestamp at which a singular transmission has been received, from the perspective of the gateway:
https://github.com/TheThingsNetwork/lorawan-stack/blob/70d88dd86a9b17bf8faf1bbea6961d6c5d3f2ae7/api/lorawan.proto#L326-L328

3. A set of timestamps of each individual transmission metadata:
https://github.com/TheThingsNetwork/lorawan-stack/blob/70d88dd86a9b17bf8faf1bbea6961d6c5d3f2ae7/api/metadata.proto#L41
https://github.com/TheThingsNetwork/lorawan-stack/blob/70d88dd86a9b17bf8faf1bbea6961d6c5d3f2ae7/api/messages.proto#L82

### Why do we need this? Who uses it, and when?

The semantics of timestamps 2 and 3 are not clear unfortunately. Specifically it is unclear what is the clock source of this timestamp (is it a GPS timestamp that the gateway is _sure_ of, or did it do some NTP and has only a bit of accuracy ?).

This has the following effects:
- Some uplinks are marked as coming from the 80s, because the gateway did some kind of NTP failure, or lacks a GPS fix.
- Some uplinks are buffered and the timestamp is mis-attributed (time of buffer submission vs actual receive timestamp).
- Correctly supporting `DeviceTimeAns` is not really possible (if we have 3 timestamps, which one do we pick ? per spec, it should be a GPS one)

### Proposed Implementation

Option 1: Change the semantics of the `time`, for uplink messages, such that the presence of the value signals that it is provided by GPS, and leave it empty otherwise.

PROS:
- It was always nullable, so taking it out is by the book correct.
- We guarantee that applications that use it have a useful timestamp.
CONS:
- De facto, this is a breaking change. Gateways that had 0 issues will suddenly no longer have a timestamp.

I don't think that we should go this way, but I'm documenting this here.

Option 2: Add a new field, `gps_time`, that is filled when the timestamp is really sourced by GPS. If we have both an NTP sourced timestamp, and a GPS sourced timestamp, `time` contains the GPS one.

PROS:
- Not a breaking change.
- It will clarify the semantics.
CONS:
- We add a new field to store. Keep in mind that this is both something we will carry around (data transfer between components) and store (event storage, NS recent uplinks storage maybe, AS storage integration), so we should be mindful.

I'm more in favor of this variant, but it's not entirely 'free' (storage / transfer costs). 

cc @johanstokking

### Contributing

- [X] I can help by doing more research.
- [X] I can help by implementing the feature after the proposal above is approved.
- [X] I can help by testing the feature before it's released.

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

adriansmares, 2022-07-11:
Another thing pointed out by https://github.com/TheThingsNetwork/lorawan-stack/issues/5586#issuecomment-1180573745 is that perhaps we should have next to the `gps_time` also the `received_at` (representing the time at which the Gateway Server has received the message), and a duration called `mean_half_rtt`.

johanstokking, 2022-07-12:
Do we even know how the gateway acquired absolute time? In which protocols is it obvious what the time source is; GPS, NTP or server-RTT based? I think we shouldn't put any gateway provided absolute time in `gps_time`. I think that the RX metadata's `time` is the best we can do, and it's up to the gateway and the protocol to provide the most accurate wall time in there.

---

What are we trying to achieve?

> * Some uplinks are marked as coming from the 80s, because the gateway did some kind of NTP failure, or lacks a GPS fix.

Isn't the issue incorrect milliseconds vs seconds Unix timestamp?

Still, if the NTP source is incorrect, what can we do?

> * Some uplinks are buffered and the timestamp is mis-attributed (time of buffer submission vs actual receive timestamp).

What can we do?

> * Correctly supporting `DeviceTimeAns` is not really possible (if we have 3 timestamps, which one do we pick ? per spec, it should be a GPS one)

Let's discuss this in #5586

adriansmares, 2022-07-12:
> Do we even know how the gateway acquired absolute time? In which protocols is it obvious what the time source is; GPS, NTP or server-RTT based?

For both UDP ([`tmms`](https://github.com/Lora-net/packet_forwarder/blob/d0226eae6e7b6bbaec6117d0d2372bf17819c438/PROTOCOL.TXT#L355)) and BasicStation ([`gpstime`](https://lora-developers.semtech.com/build/software/lora-basics/lora-basics-for-gateways/?url=tcproto.html)) we know for sure which timestamp is GPS based. The MQTT v2 format does not have this semantically specified, and MQTT v3 / Protobuf are going to automatically benefit from this clarification.

> Still, if the NTP source is incorrect, what can we do?
> What can we do?

We can specify explicitly that `gps_time` is either null (unknown), or actually a real GPS timestamp. With the current `time`, we cannot differentiate between something that is GPS sourced or just from NTP.

johanstokking, 2022-07-12:
Okay indeed if we know that the absolute time is GPS based, we can put it in `gps_time`.

But what do we do with `time` then? What is the added value of that?

adriansmares, 2022-07-12:
> But what do we do with `time` then? What is the added value of that?

In the wild, we have many gateways that are not GPS enabled, but will provide a [`tmst`](https://github.com/Lora-net/packet_forwarder/blob/d0226eae6e7b6bbaec6117d0d2372bf17819c438/PROTOCOL.TXT#L354) (in UDP) or a [`rxtime`](https://github.com/lorabasics/basicstation/blob/ba4f85d80a438a5c2b659e568cd2d0f0de08e5a7/src/s2e.c#L186) (in BasicStation). I think there is value in still allowing that to be a `time` for debugging purposes.

I also believe that there are people in the wild that now depend on a `time` being there, be it right or wrong. One of my original proposals was to say 'from now on `time` is null or GPS time', but I think that this is a breaking change that will anger many users. That ship has sailed in `3.0.0`, but we can still allow platforms to be picky about which timestamp they use.

PR #5598 (2022-07-22): Clarify the semantics of the uplink message timestamps

#### Summary

Closes #5585

#### Changes

- Added `gps_time` and `received_at` fields to `RxMetadata`
- Improved uplink message's `GatewayReceiveTime` accuracy (when possible)

#### Testing

Adjust relevant test cases to feature expected `received_at` and `gps_time` values.

##### Regressions

Possible regressions listed in #5585 and #5586.

#### Notes for Reviewers

nil

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Compatibility: The changes are backwards compatible with existing A [... cut]
closed 2022-07-22 by happyRip