TheThingsNetwork/lorawan-stack #5587: Refactor Identity Server Store interfaces

htdvisser, 2022-07-11:
### Summary

It would be good to make some changes to some Identity Server Store interfaces to make methods more "single-purpose".

### Current Situation

- The `SetMember` method of the `MembershipStore` interface deletes a collaborator when passing zero rights.
- The `UpdateAPIKey` method of the `APIKeyStore` interface deletes an API key when passing zero rights.
- The `Validate` method of the `ContactInfoStore` does multiple things that should be "business logic" instead of "store logic":
	- Getting a `ContactInfoValidation`
	- Checking for already used / expired
	- Marking `ContactInfo` as "validated"
	- Validating the primary email address of a user
	- Marking the `ContactInfoValidation` as "used"

### Desired Situation

- The `MembershipStore` interface should have separate methods for updating and deleting collaborators.
- The `APIKeyStore` interface should have separate methods for updating and deleting API keys.
- The `ContactInfoStore` should have separate methods for each of the tasks described above. Other business logic belongs in the RPC implementation.

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

htdvisser, 2022-07-11:
Blocked on #3250, so for triage let's just assign someone, and maybe schedule it for Q4 or so.

cc: @NicolasMrad

adriansmares, 2023-05-23:
What is the status here ?

I saw that we have [a branch](https://github.com/TheThingsNetwork/lorawan-stack/tree/issue/5587-refactor-is-interfaces) already, and I am wondering how much is left in order to schedule this for review and merging.

nicholaspcr, 2023-05-23:
It was supposed to be finished on the Thursday two weeks ago but didn't managed to do it. 
I'm pretty confident in finishing this Thursday.

nicholaspcr, 2023-06-12:
This was actually finished on `v3.26.1` the related RPC implementation is the continuation based on the addition of the delete methods in the interfaces.

Therefore I'll be closing this issue.

closed 2023-06-12 by nicholaspcr