TheThingsNetwork/lorawan-stack #5608: The network server does not respond to DeviceTimeRequest if the command is accompanied by a TXParamSetupAns

Achichig, 2022-07-18:
### Summary

In the case where a modem has to send a TxParamSetupAnswer and a DeviceTimeRequest, the NS sees the commands but does not answer the DeviceTimeRequest.  

### Steps to Reproduce

I saw this bug using AS923 GRP4 frequency plan, Lorawan spec 1.0.4, RP2 v1.0.3.

1. Send payload (frm_payload) 0x09 0D to NS (0x09 for TxParamSetupAns and 0x0D for DeviceTimeRequest) on port 0 (without fopts)

### Current Result

In the image below, the fcnt 6 message corresponds to TxParamSetupAns and DeviceTimeRequest (as we can see in the decrypted payload lower). We can see that following this uplink, the NS does not enqueue any downlink DeviceTimeAnswer .

[image]

[image]

### Expected Result

I am expecting the NS to answer to DeviceTimeRequest even if this mac command is combined with a TxParamSetupAnswer

### Deployment

The Things Stack Enterprise (self-hosted)

### The Things Stack Version

3.19.1

### Contributing

- [ ] I can help by doing more research.
- [ ] I can help by implementing a fix after the proposal above is approved.
- [X] I can help by testing the fix before it's released.

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

adriansmares, 2022-07-18:
Can you please provide an event stream in JSON format with verbose logging enabled ? You may share those via email (`adrian.mares@thethingsindustries.com`) if you consider that they contain sensitive data.

Note that you should never rely on having a Network Server response in the subsequent downlink - that's only the happy case. A non exhaustive list of things that can go wrong:
- The uplink containing the requests may be lost over the air or due to latency arrive too late for RX1/RX2.
- The downlink containing the response may be lost over the air or due to latency arrive too late for RX1/RX2.
- The MAC response may not fit the downlink - the NS will set the `FPending` to `true` but the end device still needs to send the unanswered requests in the subsequent uplink that opens a new receive window.

I can't tell only from the provided information why the NS did not generate a response+downlink, but I do think this bug report may be valid.

adriansmares, 2022-07-18:
Thanks to the logs - I have found the issue here. It is interesting.

---

In TTS we have the concept of pending requests - MAC command requests which are awaiting for a response from the end device. The flow is particularly easy:
1. An end device sends an uplink.
2. We walk through the MAC commands from the end device and check if the responses handle any old pending requests. These are removed from the pending requests list.
2. If there are any pending requests left, we log them in the stack Console with a warning (this should become an event instead).
3. We flush the pending requests.
4. We generate new pending requests if the desired and current state of the end device has a drift. We schedule a downlink with these requests.
5. Goto 1.

This is very much in line with what the 1.0.4 spec has in mind, with a request-response flow in which a transmitter has to re transmit a request if he wants a new response. The twist is that `RXParamSetupReq`, `RXTimingSetupReq`, `TXParamSetupReq` and `DlChannelReq` do not work like this.
[image]

The commands mentioned above actually will have the `*Ans` attached to the uplinks until the NS schedules a downlink to the end device:
[image]
[image]
[image]
[image]

---

In this issue, via private logs, I've seen that the end device keeps sending the `TxParamSetupAns` over multiple uplinks until a downlink is sent.

---

The Network Server is not really compatible right now with the fact that these `*Ans` commands will be resent multiple times. We will see an `*Ans` without a pending request and basically drop the whole set of MAC responses:
https://github.com/TheThingsNetwork/lorawan-stack/blob/5047fe7f74def34c2c42ac0516aa9b3ac323736e/pkg/networkserver/mac/mac.go#L51-L65
[(having no pending request causes us to fail on L64)
https://github.com/TheThingsNetwork/lorawan-stack/blob/v3.21/pkg/networkserver/grpc_gsns.go#L580-L583
](https://github.com/TheThingsNetwork/lorawan-stack/blob/5047fe7f74def34c2c42ac0516aa9b3ac323736e/pkg/networkserver/grpc_gsns.go#L580-L583)

---

What needs to be done here: `handleMACResponse` / `handleMACResponseBlock` should have a parameter `allowNotFound` which basically handles these 'response retransmissions'. `RXParamSetupAns`, `RXTimingSetupAns`, `TXParamSetupAns`, `DlChannelAns` should use it, while other commands should continue to disapprove unrequested responses.

johanstokking, 2022-07-19:
Yes indeed. These are sticky answers.

What do you propose will happen if there's an invalid answer? Drop the remaining MAC commands still? What does this mean for the end device state going forward?

johanstokking, 2022-07-19:
> All 3 of the above may signal the fact that the MAC state is de-synchronized, but I cannot say what we should do at that point.
> 
> Users need to be made more aware of the fact that the NS cannot communicate properly with their end device, but killing the communication completely if an anomaly is detected would be too harsh:

Yes, fully agreed. Also I think indeed for reasons 1 and 2 we'll want an event.

closed 2022-07-28 by adriansmares