TheThingsNetwork/lorawan-stack #5613: Update database schema with more NOT NULL columns

htdvisser, 2022-07-19:
### Summary

A number of columns in our database schema allow `NULL` values when in fact `NULL`s shoud not be possible. We should fix that.

### Current Situation

Many columns (at least the ones marked as `nullzero` in `pkg/identityserver/bunstore`, but there are more) allow `NULL` values that are equivalent to empty values in The Things Stack.

### Desired Situation

We should not allow `NULL`s in those columns. We can set a `DEFAULT` for `INSERT` compatibility.

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

nicholaspcr, 2023-01-10:
As pointed out by @cvetkovski98 we do have the `notnull` annotation which creates the restriction on the tables but at the same time we are not using the `nullzero` which marshals the Go zero values into NULL. Ref: https://bun.uptrace.dev/guide/models.html#struct-tags

Besides this I should write it in here what else shouldn't be NULL in the database.

nicholaspcr, 2023-09-13:
The initial idea was to find more columns which would benefit from having more NOT NULL columns. By going through columns which already have the `nullzero` restriction we can get a good idea where the addition is applicable. Hylke mentioned that there are other fields which would benefit from this addition but since the current list is extensive enough, we can develop on top of them after agreeing on the changes on the fields with `nullzero`.

I went a bit all over the place while working on this, so there are two sections in this comment:

1. The fields that have `nullzero` but would benefit from having `notnull`
2. The fields that have `notnull` but would benefit from having `nullzero`

Each section is divided in:
- What columns can have the addition
    - Which of them ensures no NULL or zero values due to request validation and business logic.
    - Which of them probably have NULL or zero values.
- What columns should not have the addition
- Some additional questions regarding the topic

Overall most fields which have `notnull` or `nullzero` either enforce the value not being null via request validation and business logic or the current behavior is adequate. But for those who require the change, imply in a different migration for each, where we would set the default value and add the constraint to the column on the table.

@adriansmares I left the queries for the fields (just counting amount of NULLs) for each fields which I believe might have the NULL or zero value on production. There are some questions regarding some specific fields which I would appreciate your take.

<details>
 <summary>Notes related to `nullzero`</summary>

### List of fields that could have `notnull` added:

###### Do not have NULL values

This means that adding the `notnull` restriction should not cause problems

- pkg/identityserver/bunstore/api_key_store.go:	APIKeyID string `bun:"api_key_id,nullzero"`
    - Uses `is.GenerateAPIKey` to fill the value, there should be no NULL value
- pkg/identityserver/bunstore/api_key_store.go:	Key    string `bun:"key,nullzero"`
    - Uses `is.GenerateAPIKey` to fill the value, there should be no NULL value
- pkg/identityserver/bunstore/api_key_store.go:	Rights []int  `bun:"rights,array,nullzero"`
    - message validation ensures at least one right, should never be NULL
- pkg/networkoperationscenter/store/model/gateway.tti.go:	DataRate  string `bun:",type:varchar(20),nullzero"`
    - Mentioned in https://github.com/TheThingsIndustries/lorawan-stack/issues/3761

- IDs have regex enforcing a value
    - pkg/identityserver/bunstore/gateway_store.go:	BrandID         string `bun:"brand_id,nullzero"`
    - pkg/identityserver/bunstore/gateway_store.go:	ModelID         string `bun:"model_id,nullzero"`
- pkg/identityserver/bunstore/end_device_location_store.go:	Service string `bun:"service,nullzero"`
    - Regex validation makes it so its always non empty
- pkg/identityserver/bunstore/contact_info_store.go:	Reference string `bun:"reference,nullzero"`
    - Regex validation makes it so its always non empty
    - Reference is the ID
- pkg/identityserver/bunstore/contact_info_store.go:	Token     string `bun:"token,nullzero"`
    - Regex validation makes it so its always non empty
- pkg/identityserver/bunstore/notification_store.go:	Receivers []int `bun:"receivers,array,nullzero"`
    - validation makes it so its always non empty

###### Might HAVE NULL values

This means that adding the `notnull` restriction requires checking on prod if it is truly the case and setting a default if so.

- pkg/networkoperationscenter/store/model/end_device.tti.go:	SessionKeyID []byte `bun:",type:bytea,nullzero"`
    - It should not but it might have a NULL value since there is no validation for the field.
- pkg/identityserver/bunstore/user_session_store.go:	SessionSecret string `bun:"session_secret,nullzero"`
    - No validation on making it necessary or specifying minimum value
    - It says it will never be returned but `FindSessions` does return a list of sessions with the secret included.
        - `// The session secret is used to compose an authorization key and is never returned.`
- pkg/identityserver/bunstore/client_store.go:	ClientSecret string `bun:"client_secret,nullzero"`
    - No validation on making it necessary or specifying minimum value
- pkg/identityserver/bunstore/gateway_store.go:	GatewayServerAddress string `bun:"gateway_server_address,nullzero"`
    - There is validation but was introduced on v3.20, so might have NULL values
- pkg/identityserver/bunstore/application_store.go:	NetworkServerAddress     string `bun:"network_server_address,nullzero"`
    - There is validation but was introduced on v3.20, so might have NULL values
- pkg/identityserver/bunstore/application_store.go:	ApplicationServerAddress string `bun:"application_server_address,nullzero"`
    - There is validation but was introduced on v3.20, so might have NULL values
- pkg/identityserver/bunstore/application_store.go:	JoinServerAddress        string `bun:"join_server_address,nullzero"`
    - There is validation but was introduced on v3.20, so might have NULL values

[... 272 lines cut ...]
- identityserver/bunstore/gateway_store.go:                        EnforceDutyCycle                 `bun:"enforce_duty_cycle,notnull"`
- identityserver/bunstore/gateway_store.go:                        UpdateLocationFromStatus         `bun:"update_location_from_status,notnull"`
[... cut ...]
And lastly these lists are useful for asking questions regarding specific fields but their main purpose is also to serve as a record of things to change. This way the issue can be fragmented into smaller PRs. While it these changes have to be merged on a minor release I believe it is important to not attempt to do everything at once, specially considering the overhead generated by a mistake on the identification in one of these columns behavior. Will break the list into sections done by minor releases but I'm posting this in order to have a discussion around the questions that appeared.

nicholaspcr, 2023-10-18:
This requires a migration, therefore I'm moving it to the `v3.29.0`

KrishnaIyer, 2024-06-27:
@nicholaspcr: This is completed right?

nicholaspcr, 2024-06-27:
I don't think it is, I'll double check the places where I previously marked that needed changes and update this issue