TheThingsNetwork/lorawan-stack #5616: Support ordering in List*Collaborator requests

htdvisser, 2022-07-19:
### Summary

It would be nice if we could support custom ordering in `List*Collaborator` requests.

### Current Situation

We support pagination, but ordering is always by account ID (I think)

### Why do we need this? Who uses it, and when?

I remember @kschiffer mentioned that this would be nice to have because of consistency with other listings in the Console. Can't find the issue though.

### Proposed Implementation

- Add field to `List*Collaborator` requests in the API
- Pass it to the IS store (using context)
- Support it in the IS store

### Contributing

- [ ] I can help by doing more research.
- [ ] I can help by implementing the feature after the proposal above is approved.
- [ ] I can help by testing the feature before it's released.

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

kschiffer, 2022-07-19:
It would indeed be more consistent with other entities. I added a lot of sorting functionality via https://github.com/TheThingsNetwork/lorawan-stack/pull/5497 but had to implement a workaround for sorting collaborators in the frontend because the backend doesn't allow it.

nicholaspcr, 2022-09-29:
Changes the assignees due to the backend section of this being done. 
Tagged @kschiffer in hopes that he can best assign the issue to the appropriate person.