TheThingsNetwork/lorawan-stack #5643: Promote bunstore and remove gormstore

htdvisser, 2022-07-26:
### Summary

https://github.com/TheThingsNetwork/lorawan-stack/pull/5642 added a new implementation of the Identity Server storage layer (`pkg/identityserver/bunstore`) behind a feature flag (`is.bunstore`). Once we're satisfied with the new implementation, we should...

1. Enable the feature flag by default, still allowing it to be disabled in case of unexpected behavior.
2. Remove the feature flag and the old implementation (`pkg/identityserver/gormstore`).

This issue replaces https://github.com/TheThingsNetwork/lorawan-stack/issues/3250

### Current Situation

- https://github.com/TheThingsNetwork/lorawan-stack/pull/5642 added a new implementation of the Identity Server storage layer (`pkg/identityserver/bunstore`) behind a feature flag (`is.bunstore`).
- The old implementation (`pkg/identityserver/gormstore`) is still used by default
- The store tests (`pkg/identityserver/storetest`) still have some TODOs about un-commenting tests that fail on the old implementation, but should pass on the new implementation:
	- https://github.com/TheThingsIndustries/lorawan-stack/issues/3033
	- https://github.com/TheThingsNetwork/lorawan-stack/issues/5046
	- https://github.com/TheThingsIndustries/lorawan-stack/issues/3034
	- I'll close those issues in favor of this one.

### Desired Situation

The old implementation should be gone once the new implementation is promoted to "stable".

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

nicholaspcr, 2022-09-13:
Writing here that it was agreed on 13/09/22 meeting that on the v3.22.0, the bunstore should be set as the default store and there should be an option to enable the gormstore. Basically invert the current situation where bunstore is only used if explicitly set through the experimental flags.

After a certain amount of time drop the gormstore implementation. Using this issue

nicholaspcr, 2022-10-06:
Keeping this updated for interested parties.
This is being tested on TTN and making it the default store was postponed due to the following bugs: #5843, #5830.

nicholaspcr, 2022-11-04:
Updating this issue.

The bunstore was set as the default implementation on #5897 but there is still the need to remove the `gormstore` implementation from the codebase.

nicholaspcr, 2023-01-17:
Closing this since the `gormstore` was removed in https://github.com/TheThingsNetwork/lorawan-stack/pull/6009

closed 2023-01-17 by nicholaspcr