TheThingsNetwork/lorawan-stack #5691: Add OpenTelemetry Tracing

htdvisser, 2022-08-10:
### Summary

In https://github.com/TheThingsNetwork/lorawan-stack/pull/5642 I started adding OpenTelemetry trace regions to `pkg/identityserver/bunstore`. We should instrument more code in other packages, and set up exporters and such.

### Current Situation

We have some OpenTelemetry instrumentation in `pkg/identityserver/bunstore`, but the rest of the OpenTelemetry SDK has not been set up.

### Why do we need this? Who uses it, and when?

OpenTelemetry tracing is going to be useful for getting more insights into how messages pass trough The Things Stack and can help us understand how one component affects others, especially as the system grows.

### Proposed Implementation

- Instrument more code with OpenTelemetry trace spans ([docs](https://opentelemetry.io/docs/instrumentation/go/getting-started/#trace-instrumentation))
- Set up the OpenTelemetry SDK ([docs](https://opentelemetry.io/docs/instrumentation/go/getting-started/#sdk-installation))

### Contributing

- [ ] I can help by doing more research.
- [ ] I can help by implementing the feature after the proposal above is approved.
- [ ] I can help by testing the feature before it's released.

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

adriansmares, 2022-11-29:
@happyRip you may start looking into this. I think we shouldn't do everything in one big PR but rather you can consider adding these to small packages initially (`pkg/crypto` for example) and then continue with components (`pkg/applicationserver`, `pkg/gatewayserver` and so forth).

As an initial step I recommend looking into how we can setup the SDK, then we can look at the instrumentation of `pkg/identityserver/bunstore` and continue with adding this to other packages.

closed 2023-03-09 by happyRip

htdvisser, 2023-03-09:
Nice 😎

PR #6004 (2023-03-09): Add OpenTelemetry Tracing

#### Summary

Closes #5691 

#### Changes

- Add telemetry configuration 
- propagate telemetry in component rpc context

#### Testing

To test this PR first an instance of `jaeger` is needed ([Jaeger All in One](https://www.jaegertracing.io/docs/1.41/getting-started/#all-in-one)).
After running `jaeger` access the UI at http://localhost:16686.

Then build and run `lorawan-stack` from `feature/opentelemetry-tracing` branch.
To use the telemetry feature a couple flags need to be set:

```bash
go run ./cmd/ttn-lw-stack -c ./config/stack/ttn-lw-stack.yml start \
  --tra [... cut]