TheThingsNetwork/lorawan-stack #5702: `version_ids.serial_number` field is ignored in the device creation API call

ymgupta, 2022-08-11:
### Summary

While creating the device using APIs, we see that the `version_ids.serial_number` field is ignored. We have included the `version_ids` values (`brand_id`, `model_id`, `serial_number`) in the request body of Identity Server API call and field mask. But, we see that only `brand_id`, and `model_id` are stored for the device. The API response doesn't contain the `serial_number` field. However, as per the [API reference documentation](https://www.thethingsindustries.com/docs/reference/api/end_device/#message:EndDeviceVersionIdentifiers), we see that `serial_number` is part of the version identifiers. And, it is stored in Entity Registry, Network Server and Application Server.

### Steps to Reproduce

1. Create a device in all components of TTS using APIs.
2. In the request body of the API call to IS, NS and AS, set the `version_ids` as follows:
```
"version_ids": {
            "brand_id": "the-things-products",
            "model_id": "the-things-uno",
            "serial_number": "some-sample-serial-number"
        }
```
Also, set the following values in the field mask of the API call:
```
"version_ids.brand_id",
"version_ids.model_id",
"version_ids.serial_number"
```

### Current Result

1. `brand_id` and `model_id` are stored successfully for the device with the IS API call, but the `serial_number` is ignored.
2. The IS API response doesn't contain the `serial_number`.
3. Also, the NS and AS doesn't accept `serial_number` in the field mask. Including it in the API calls raises `forbidden path` error.
4. Additionally, we don't see the flag to create the `version_ids.serial_number` in CLI as well.
```
$ ttn-lw-cli device create --help | grep version-ids
     
      --version-ids.band-id string                                                            
      --version-ids.brand-id string                                                           
      --version-ids.firmware-version string                                                   
      --version-ids.hardware-version string                                                   
      --version-ids.model-id string    
```

### Expected Result

The IS, AS and NS should store the `serial_number` value from the API requests.

### Relevant Logs

```shell
## 1. Identity Server:

**Request:**

curl --location --request POST 'https://<tenant-id-redacted>.eu1.cloud.thethings.industries/api/v3/applications/<app-id-redacted>/devices' \
--header 'authority: <tenant-id-redacted>.eu1.cloud.thethings.industries' \
--header 'accept: application/json, text/plain, */*' \
--header 'accept-language: en-GB,en-US;q=0.9,en;q=0.8' \
--header 'authorization: Bearer XXXXXXXXXX' \
--header 'content-type: application/json' \
--data-raw '{
    "end_device": {
        "ids": {
            "device_id": "things-uno-1",
            "dev_eui": "<dev-eui-redacted>",
            "join_eui": "0000000000000000",
            "application_ids": {
                "application_id": "<app-id-redacted>"
            }
        },
        "join_server_address": "<tenant-id-redacted>.eu1.cloud.thethings.industries",
        "network_server_address": "<tenant-id-redacted>.eu1.cloud.thethings.industries",
        "application_server_address": "<tenant-id-redacted>.eu1.cloud.thethings.industries",
        "version_ids": {
            "brand_id": "the-things-products",
[... 292 lines cut ...]

v3.21.0

### Client Name and Version

```shell
Using the CLI: ttn-lw-cli 3.21.0 on Ubuntu 18.04.6 LTS
Using cURL: curl 7.58.0 (x86_64-pc-linux-gnu)
```

### Other Information

Currently, the details such as `brand_id` and `model_id` are available in the `version_ids` object of uplink messages. They are used to route the messages in the user's application based on those values. However, the user wants to route the messages based on the `serial_number` value as well.

Ref: https://github.com/TheThingsIndustries/lorawan-stack-support/issues/821

### Contributing

- [ ] I can help by doing more research.
- [ ] I can help by implementing a fix after the proposal above is approved.
- [ ] I can help by testing the fix before it's released.

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

johanstokking, 2022-08-18:
Honestly I think that we made a mistake by introducing https://github.com/TheThingsNetwork/lorawan-stack/blob/v3.21/api/identifiers.proto#L163-L167 to `EndDeviceVersionIdentifiers`. As you say, these aren't version identifiers.

The serial number is metadata of the end device. It should be a toplevel field of `EndDevice`. And shouldn't go in upstream messages.

The Vendor and Profile IDs are something typical TR005.

@KrishnaIyer can we deprecate these fields based on progressive insight?

KrishnaIyer, 2022-08-18:
Ok I agree indeed, we can deprecate it. I'll add a quick PR

PR #5971 (2022-12-20): Move `serial_number`, `vendor_id` and `vendor_profile_id` to `EndDevice` root
#### Summary
This PR moves the `serial_number`, `vendor_id` and `vendor_profile_id` fields from the nested `version_ids` to the root of `EndDevice` with necessary changes.

Closes #5702 

#### Changes
- Introduced a new migration to add the `serial_number` column to the IS database.
- Updated the GORM and Bun models to reflect the new column.
- Updated the tests to reflect the new column.
- `EndDeviceVersionIdentifiers` proto message no longer has `serial_number`, `vendor_id` and `vendor_profile_id` fields.
- `EndDev [... cut]
closed 2023-01-10 by NicolasMrad