TheThingsNetwork/lorawan-stack #5710: Support entity tags

kschiffer, 2022-08-17:
### Summary

As a means of logically grouping entities and assigning different treatments or performing batch actions based on such groups, we should introduce entity tags.

### Current Situation

There is no way of grouping entities other than maybe assigning IDs with a prefix, which is only an implicit way of grouping.

### Why do we need this? Who uses it, and when?

As stated above, tags are a way of grouping entities. Such grouping has several benefits:

- Enabling explicit visual grouping as a means of organizing an entity fleet logically and facilitate device fleet management.
- Improving entity search.
- Assigning different settings to groups of end devices based on tags. For e.g. we could make integrations assignable by tags so that multiple integrations can be set up for different slices of an application's end devices.
- Selecting devices for batch operations, such as moving devices between applications or deleting end devices in batch.

### Proposed Implementation

This issue should be scoped to only the tags themselves and their assignment to entities.

We would need a new tag model which contains information about the tag such as:
- ID
- Name
- Description
- Color

Then we need to add an array field to the entities `tags` which contains the list of assigned tag IDs.

To assign the tags, we would require an API that allows adding/removing a list of tags to/from a list of entities.

Then eventually, we need to update our list/search APIs to include filtering by tag IDs.

### Contributing

- [X] I can help by doing more research.
- [X] I can help by implementing the feature after the proposal above is approved.
- [X] I can help by testing the feature before it's released.

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

johanstokking, 2022-08-17:
As discussed offline, this would work very much like GitHub labels. This would indeed not just be for end devices, but for other entities as well, most notably gateways and applications.

This is different from attributes in the sense that this is a list of independent values, and we should efficiently filter on the presence of a value. Since TTS would have first class support for this, it wouldn't be an attribute anyway, as those are custom user defined things.

pierrephz, 2025-03-06:
Here is how the label could work on the console:

# Label page
- List of labels with `Name`, `Description`, `Related entities` (how many applications, gateways and end devices have this label), and `Actions` (Edit and Delete)

[image]

[image]

## How to create a label?
- From the label page, same interaction like on Github; Click on `Create label`, it opens a label creation panel.
- From the entity creation page, add a `End device propriety` section to allow the user to assign or create a label from there.

[image]

[image]

## How to access and use a label?

### From the Home page
- Add a label interaction in the Header.
- Like with the Notification interaction; display the most recent labels with the number of related entities.
- Add a link at the bottom of the dropdown to the label page to see all of them.

[image]

### With the Search interaction
- Add a nested input below the main search input to add an extra filter with label on the search. Having a nested input below the main one allows us to keep a hierarchy and keep the main search input in the center of the attention.
- When opened, the nested label input also have a dropdown to allow the user to select a label from a list, without having to type it. It is an important feature since the user might not know every label by heart.

[image]

[image]

[image]

### In a entity page
- The label is visible next to the name of the entity at the top of the page.

[image]

@johanstokking @KrishnaIyer this is a proposal on how to use the labels in the console, let me know if you have feedback or if you think something is missing, then we can work with @ryaplots on the implementation

ryaplots, 2025-03-07:
>     1. This only shows adding labels to new entities. How can users assign them to exisiting ones?
True! We forgot to discuss how to assign labels to existing entities or how to edit labels. It should be possible to open the overview of any entity and add/edit the labels

pierrephz, 2025-03-07:
> 2. For my untrained UX eyes, I'm not so sure of the extra roundedness (that's not a real word). We don't have any other element with this much rounded corners and to my eyes it looks odd.

It make sense that we don't have any other element with this roundness, since we are introducing a new feature (labels).
Having them "square" like our button would make them too similar, but they serve distinct purposes (categorisation for the labels / actions for the buttons) so they have to be different.

I was using the same UI as Github there, but if you think they are too catchy with the background colors, I can look into a more subtle UI, like color dot + text and color borders

pierrephz, 2025-03-07:
> 1. This only shows adding labels to new entities. How can users assign them to exisiting ones?

Here is a proposal to add assign a label to en existing entity (from the entity page):
- Add a clickable label placeholder (+ Add label) in the labels' spot next to the entity name. When they are already some labels, add this placeholder at the end of the labels' raw.
- Clicking on it will open a modal where you can add an existing label, or create a new one.

[image]

[image]

[image]

KrishnaIyer, 2025-03-10:
> Having them "square" like our button would make them too similar, but they serve distinct purposes (categorisation for the labels / actions for the buttons) so they have to be different.

Yeah, you're probably right. Let's stick to this for now and see how it works out. Thanks!

pierrephz, 2025-03-10:
> We also need, in each entity list, a way to filter by label. [@pierrephz](https://github.com/pierrephz)

Here is a way to filter entities by labels:
- Add a `Label` filter next to the search bar
- On click, open a dropdown list where you can search, select labels, and clear the current selection
- If some labels are selected, the filter gets "active" (black) to show user some labels are being applied

[image]

[image]

[image]

[image]

Let me know if you have any thoughts

johanstokking, 2025-03-13:
Just chiming in here that this is gonna be really great.

KrishnaIyer, 2025-04-14:
Tested in https://github.com/TheThingsIndustries/lorawan-stack-support/issues/1267 and can be closed.

closed 2025-04-14 by KrishnaIyer