TheThingsNetwork/lorawan-stack #5751: Abort redux request actions only when necessary

kschiffer, 2022-08-31:
### Summary

The webui currently aborts any pending request actions whenever a route change is performed. This can lead to situations where requests are aborted without replacement, if a route change does not trigger a rerender of the `withRequest()`-hoc or `<Request />` component. This happens when the route change only changes children of the nodes that do the necessary request.

### Steps to Reproduce

1. Go to end device overview
2. Before the consequent request actions have resolved, navigate to subpage, e.g. API Keys
3. The launched requests will hence be aborted
4. Go back to the device overview
5. Observe an error

### Current Result

Console crashing

### Expected Result

The Console should only abort the requests that became obsolete due to the route change, any other requests should not be aborted until they became obsolete.

### Deployment

The Things Stack Cloud

### Proposed Fix

The resolution here has multiple layers

- We currently still use the `withRequest()`-hoc in many places, these should be ideally replaced with `<Request />`
- We can use `componentDidMount()` and `componentWillUnmount()` to keep track of pending requests for the current subtree and to abort those pending requests when unmounting
- For that to work, we need to have access to the action type within `withRequest`. However, we currently do not always return the dispatch in the passed `mapPropsToDispatch`. It is however necessary to return the result of `dispatch` to get access to the dispatched action.
- It's hence necessary to refactor the usage of `withRequest()` to do that. Alternatively we can refactor these ocurrences to use `<Request />`

### Contributing

- [X] I can help by doing more research.
- [X] I can help by implementing a fix after the proposal above is approved.
- [X] I can help by testing the fix before it's released.

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

adriansmares, 2023-05-23:
I saw some discussions regarding `withRequests` recently - should we get a concrete list of current usages and start distributing these between people for the refactor ?

KrishnaIyer, 2023-07-04:
Let's make this actionable this month and then assign it for picking up.

kschiffer, 2023-08-17:
We've indeed completely removed the use of `withRequest()` in favor of `<RequireRequest />` recently. This should make resolving this issue more straightforward.

The way I see this being resolved is keeping track of the currently relevant pending request actions in an object of arrays within `<WithRequest />` in the shape of:
```js
{
  "/route/to/view": ["RequestAction1", "RequestAction2"],
  "/route/to/view/sub": ["RequestAction3", "RequestAction4"]
}
```
Then, when changing the route fire only the respective abort actions of unmounted route.

I might be overthinking this and there could be a simpler solution maybe making use of some `ReactRouter`-hooks, but I'd need to look further into that.

KrishnaIyer, 2024-10-14:
@mjamescompton: Is this still relevant?

mjamescompton, 2024-10-15:
Unable to to recreate the issue so closing

closed 2024-10-15 by mjamescompton