TheThingsNetwork/lorawan-stack #5758: Show normalized payload in the Console

johanstokking, 2022-09-01:
### Summary

Show normalized payload in the Console

### Current Situation

The new `normalized_payload` field in `ApplicationUplink` and `ApplicationNormalizedUplink` contain a known data schema: [JSON Schema](https://github.com/TheThingsNetwork/lorawan-devices/blob/master/lib/payload.json) and [documentation](https://www.thethingsindustries.com/docs/integrations/payload-formatters/javascript/uplink/#normalized-payload).

There is now very basic support for showing `air` and `wind` (https://github.com/TheThingsNetwork/lorawan-stack/commit/425b41a29ef38ecbbc2695abb35f48c34d4434c2), showing values as JSON.

The payload code editor shows `normalized_payload{_warnings}` as returned fields, but not as prominently as decoded uplink data.

### Why do we need this? Who uses it, and when?

I think it would be good UX to show normalized payload if it's available.

### Proposed Implementation

Low hanging fruit would be to show the normalized payload nicely in the payload editor. This would need it's own code editor. If the normalized payload is present, it might be shown more prominent than decoded payload (but one does not replace the other).

What would be nice in the event view, is to show the values nicely. For example: `Air: 25.4 ℃, wind: 4 bft (NNE)` (if the `normalized_payload` were to contain `{"air":{"temperature":25.4},"wind":{"speed":6.1,"direction":15}}`.

Cherry on the pie would be to render temperature graphs based on historical data retrieved from the storage integration. We still don't want to be come an IoT platform, so this is a sliding edge, but it would be cool.

### Contributing

- [X] I can help by doing more research.
- [ ] I can help by implementing the feature after the proposal above is approved.
- [X] I can help by testing the feature before it's released.

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

KrishnaIyer, 2023-07-04:
Also good to atleast have the wireframes for this early in Q3.

kschiffer, 2023-08-31:
I've finished designing this feature as part of the general Console UX updates. Planning will follow.

KrishnaIyer, 2024-10-06:
@kschiffer Do we have wireframes for this? If not, can we work on that with @pierrephz?

@mjamescompton for the implementation.

kschiffer, 2024-10-10:
I was skimming through my files and only found two screendesigns based on an earlier revision of the Console redesign. I will discuss this with @pierrephz so he can pick this up and finalize the UX.

pierrephz, 2024-10-21:
Here is a mockup with a first proposal to display `Normalized payload` in the console:

[image]

### Overview:

[image]

johanstokking, 2024-10-21:
I think this is a good place to do it and it stays clean.

I think having a table like this is too limiting though. I think we should create sections for the high level objects that we have: battery, soil, air, wind, water, etc.

pierrephz, 2024-10-25:
I have gathered all our `properties items` from [here](https://github.com/TheThingsNetwork/lorawan-devices/blob/master/lib/payload.json#L53) and updated my mockup based on them:

### Here is the propriety element that I designed:

[image]

### Here is the updated panel with the normalized payload properties in use:

[image]

KrishnaIyer, 2025-07-14:
This is completed in https://github.com/TheThingsNetwork/lorawan-stack/issues/5758

closed 2025-07-14 by KrishnaIyer

KrishnaIyer, 2025-07-14:
This is tested and completed.