TheThingsNetwork/lorawan-stack #5812: Bug: device search using CLI

vkrasnhak, 2022-09-19:
### Summary

Device search using CLI is ignoring --dev-eui-constains flag

### Steps to Reproduce

1. Try to search a device using --dev-eui-constains flag in CLI. For example, run `ttn-lw-cli.exe end-devices search application-id-x --dev-eui-contains AAAAAAAAAAAAAAAA`

### Current Result

[{
  "ids": {
    "device_id": "000001",
    "application_ids": {
      "application_id": "application-id-x"
    },
    "dev_eui": "AAAAAAAAAAAAAAAA",
    "join_eui": "07F7277771540D16"
  },
  "created_at": "2021-10-25T18:52:15.367Z",
  "updated_at": "2021-10-26T13:27:19.606Z"
}, {
  "ids": {
    "device_id": "000002",
    "application_ids": {
      "application_id": "application-id-x"
    },
    "dev_eui": "BBBBBBBBBBBBBBBB",
    "join_eui": "07F7277771540D16"
  },
  "created_at": "2021-10-25T18:52:30.709Z",
  "updated_at": "2021-11-30T12:41:00.851Z"
}, {
  "ids": {
    "device_id": "000003",
    "application_ids": {
      "application_id": "application-id-x"
    },
    "dev_eui": "CCCCCCCCCCCCCCCC",
    "join_eui": "07F7277771540D16"
  },
  "created_at": "2021-10-25T18:52:46.166Z",
  "updated_at": "2021-12-01T17:53:54.276Z"
}, {
  "ids": {
    "device_id": "000004",
    "application_ids": {
      "application_id": "application-id-x"
    },
    "dev_eui": "DDDDDDDDDDDDDDDD",
    "join_eui": "07F7277771540D16"
  },
  "created_at": "2021-11-08T18:48:30.291Z",
  "updated_at": "2021-11-30T14:41:51.385Z"
}]

### Expected Result

[{
  "ids": {
    "device_id": "000001",
    "application_ids": {
      "application_id": "application-id-x"
    },
    "dev_eui": "AAAAAAAAAAAAAAAA",
    "join_eui": "07F7277771540D16"
  },
  "created_at": "2021-10-25T18:52:15.367Z",
  "updated_at": "2021-10-26T13:27:19.606Z"
}]

### Deployment

The Things Stack Cloud

### The Things Stack Version

3.21.2

### Client Name and Version

```shell
The Things Network Command-line Interface: ttn-lw-cli
Version:             3.17.1
Build date:          2022-01-12T12:16:36Z
Git commit:          10da1157c
Go version:          go1.17.5
OS/Arch:             windows/amd64
```

### Other Information

This bug started after the 5th of September

### Contributing

- [ ] I can help by doing more research.
- [ ] I can help by implementing a fix after the proposal above is approved.
- [ ] I can help by testing the fix before it's released.

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

nicholaspcr, 2022-09-30:
Hello @vkrasnhak, I tried replicating the issue and I wasn't able to do so. I'm hoping that you could share a bit more of the details in order to facilitate me reproducing this.

For more details on how I went about testing this:
- Ran TTS on the `v3.21.2` as mentioned.
- Executed the following commands on the `v3.17.1` CLI:
```bash
# login
go run ./cmd/ttn-lw-cli login

# create test application (test-app)
go run ./cmd/ttn-lw-cli app create --application-id test-app --user-id admin

# create three devices with json file
go run ./cmd/ttn-lw-cli dev create --application-id test-app < .env/tests/devs.json

# List devices just to check
go run ./cmd/ttn-lw-cli dev list test-app

# search by eui
go run ./cmd/ttn-lw-cli end-devices search test-app --dev-eui-contains AAAAAAAAAAAAAAAA
```
In the `.env/tests/devs.json` is pretty much three mock devices. Reference file: [devs.txt](https://github.com/TheThingsNetwork/lorawan-stack/files/9683730/devs.txt)

I got what you described in the desired result:
```
[{
  "ids": {
    "device_id": "dev-1",
    "application_ids": {
      "application_id": "test-app"
    },
    "dev_eui": "AAAAAAAAAAAAAAAA",
    "join_eui": "0000000000000000"
  },
  "created_at": "2022-09-30T11:22:26.350Z",
  "updated_at": "2022-09-30T11:22:26.350Z"
}]
```

vkrasnhak, 2022-09-30:
Hi @nicholaspcr, I am using The Things Stack Cloud with tenant-id furukawa and region eu1.

I tried the commands and the result was the same. I wasn't able to filter the devices using the --dev-eui-constains flag.

### CLI version
$ ttn-lw-cli.exe version
```
The Things Network Command-line Interface: ttn-lw-cli
Version:             3.17.1
Build date:          2022-01-12T12:16:36Z
Git commit:          10da1157c
Go version:          go1.17.5
OS/Arch:             windows/amd64
```

### Create test application (test-app)
$ ttn-lw-cli.exe app create --application-id test-app --user-id xxxxxx
```
{
  "ids": {
    "application_id": "test-app"
  },
  "created_at": "2022-09-30T12:17:51.451840Z",
  "updated_at": "2022-09-30T12:17:51.451841Z",
  "administrative_contact": {
    "user_ids": {
      "user_id": "xxxxxx"
    }
  },
  "technical_contact": {
    "user_ids": {
      "user_id": "xxxxxx"
    }
  }
}
```
### Create three devices
$ ttn-lw-cli.exe dev create --application-id test-app < devs.txt
```
{
  "ids": {
    "device_id": "dev-1",
    "application_ids": {
      "application_id": "test-app"
    },
    "dev_eui": "AAAAAAAAAAAAAAAA",
    "join_eui": "AAAAAAAABBBBBBBB"
  },
  "created_at": "2022-09-30T12:18:47.574514Z",
  "updated_at": "2022-09-30T12:18:48.482994596Z",
  "attributes": {},
  "version_ids": {
    "brand_id": "laird",
    "model_id": "rs1xx-ext-multi-sensor",
    "hardware_version": "rev 4",
    "firmware_version": "6.1_20_6_16",
    "band_id": "EU_863_870"
  },
  "network_server_address": "furukawa.eu1.cloud.thethings.industries",
  "application_server_address": "furukawa.eu1.cloud.thethings.industries",
  "join_server_address": "furukawa.eu1.cloud.thethings.industries",
  "lorawan_version": "MAC_V1_0_2",
  "lorawan_phy_version": "PHY_V1_0_2_REV_B",
  "frequency_plan_id": "EU_863_870_TTN",
  "supports_join": true,
  "root_keys": {
    "app_key": {
      "key": "FF29540EA12D453F2A3E5E360A665BF4"
    }
  },
  "mac_settings": {
    "supports_32_bit_f_cnt": true
  },
  "formatters": {
    "up_formatter": "FORMATTER_REPOSITORY",
    "down_formatter": "FORMATTER_REPOSITORY"
  }
}
{
  "ids": {
    "device_id": "dev-2",
    "application_ids": {
      "application_id": "test-app"
    },
    "dev_eui": "BBBBBBBBBBBBBBBB",
    "join_eui": "AAAAAAAABBBBBBBB"
  },
  "created_at": "2022-09-30T12:18:48.797319Z",
  "updated_at": "2022-09-30T12:18:49.699315458Z",
  "attributes": {},
  "version_ids": {
    "brand_id": "laird",
    "model_id": "rs1xx-ext-multi-sensor",
    "hardware_version": "rev 4",
    "firmware_version": "6.1_20_6_16",
    "band_id": "EU_863_870"
  },
  "network_server_address": "furukawa.eu1.cloud.thethings.industries",
  "application_server_address": "furukawa.eu1.cloud.thethings.industries",
  "join_server_address": "furukawa.eu1.cloud.thethings.industries",
  "lorawan_version": "MAC_V1_0_2",
  "lorawan_phy_version": "PHY_V1_0_2_REV_B",
  "frequency_plan_id": "EU_863_870_TTN",
  "supports_join": true,
  "root_keys": {
    "app_key": {
      "key": "FF29540EA12D453F2A3E5E360A665BF4"
    }
  },
  "mac_settings": {
    "supports_32_bit_f_cnt": true
  },
  "formatters": {
    "up_formatter": "FORMATTER_REPOSITORY",
    "down_formatter": "FORMATTER_REPOSITORY"
  }
}
{
  "ids": {
    "device_id": "dev-3",
    "application_ids": {
      "application_id": "test-app"
    },
    "dev_eui": "CCCCCCCCCCCCCCCC",
    "join_eui": "AAAAAAAABBBBBBBB"
  },
  "created_at": "2022-09-30T12:18:49.922023Z",
  "updated_at": "2022-09-30T12:18:50.715417519Z",
  "attributes": {},
  "version_ids": {
    "brand_id": "laird",
    "model_id": "rs1xx-ext-multi-sensor",
    "hardware_version": "rev 4",
    "firmware_version": "6.1_20_6_16",
    "band_id": "EU_863_870"
  },
  "network_server_address": "furukawa.eu1.cloud.thethings.industries",
  "application_server_address": "furukawa.eu1.cloud.thethings.industries",
  "join_server_address": "furukawa.eu1.cloud.thethings.industries",
  "lorawan_version": "MAC_V1_0_2",
  "lorawan_phy_version": "PHY_V1_0_2_REV_B",
  "frequency_plan_id": "EU_863_870_TTN",
  "supports_join": true,
  "root_keys": {
    "app_key": {
      "key": "FF29540EA12D453F2A3E5E360A665BF4"
    }
  },
  "mac_settings": {
    "supports_32_bit_f_cnt": true
  },
  "formatters": {
    "up_formatter": "FORMATTER_REPOSITORY",
    "down_formatter": "FORMATTER_REPOSITORY"
  }
}
```

### Search by eui
$ ttn-lw-cli.exe end-devices search test-app --dev-eui-contains AAAAAAAAAAAAAAAA
```
[{
  "ids": {
    "device_id": "dev-1",
    "application_ids": {
      "application_id": "test-app"
    },
    "dev_eui": "AAAAAAAAAAAAAAAA",
    "join_eui": "AAAAAAAABBBBBBBB"
  },
  "created_at": "2022-09-30T12:18:47.574514Z",
  "updated_at": "2022-09-30T12:18:47.574514Z"
}, {
  "ids": {
    "device_id": "dev-2",
    "application_ids": {
      "application_id": "test-app"
    },
    "dev_eui": "BBBBBBBBBBBBBBBB",
    "join_eui": "AAAAAAAABBBBBBBB"
  },
  "created_at": "2022-09-30T12:18:48.797319Z",
  "updated_at": "2022-09-30T12:18:48.797319Z"
}, {
  "ids": {
    "device_id": "dev-3",
    "application_ids": {
      "application_id": "test-app"
    },
    "dev_eui": "CCCCCCCCCCCCCCCC",
    "join_eui": "AAAAAAAABBBBBBBB"
  },
  "created_at": "2022-09-30T12:18:49.922023Z",
  "updated_at": "2022-09-30T12:18:49.922023Z"
}]
```

closed 2022-10-06 by nicholaspcr