TheThingsNetwork/lorawan-stack #5817: Allow mounting /api/v3 in subpath

RasmusGodske, 2022-09-21:
### Summary

Allow to mount the /api/v3 endpoint in a subpath, which currently does not seem to be possible(Correct me if I'm wrong). 

### Current Situation

Currently it is possible to configure the mount path of:
- console component by configuring `console.mount`
- application server by configuring `as.mount`

I'm not entirely sure sure which stack component, serves the /api endpoints, however configuring the above mentioned options, does not seem to apply to this endpoint.

### Why do we need this? Who uses it, and when?

This feature is useful for organising larger systems with multiple applications on the same domain by using subpaths. This would be possible by routing the requests using a reverse proxy such as Traefik. However from my understand the lorawan-stack has to support this, by mounting the stack under a specific subpath.

My personal use case is it host the-things-stack in a kubernetes environment, which has a single IP and a single domain pointing which is shared between multiple services eg:
- Grafana: `/admin/grafana`
- NodeRed: `/admin/nodered`
- The Things Stack: `/tts`

Without using subpath I need to create a whole new subdomain, and route depending on the host instead of the path prefix. This can also give more work, by having to create new certificates for that specific subdomain.

### Proposed Implementation

Adding a mount option for the service which serves the /api endpoint(Not sure which one it is).

### Contributing

- [X] I can help by doing more research.
- [ ] I can help by implementing the feature after the proposal above is approved.
- [X] I can help by testing the feature before it's released.

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

KrishnaIyer, 2022-09-27:
The design of the standard hosting models for the The Things Stack assumes that the API routes (and LoRa Basics Station routes that do not use `/api/v3`) are found at the root. 
So we will break a lot of assumptions in our deployment models by adding a configurable prefix.
To achieve what you intend, you can use a subdomain for your admin/node-red paths.

RasmusGodske, 2022-09-29:
I'm sad to hear that, would have loved this feature, it would have made deployment easier in some cases, but I understand.

Yes I am aware of that I could create subdomains for some of the services, however the purpose of that was to completely avoid going down the subdomain route.

Maybe document this limitation could be useful for others in the future, just so they do not have to go down the same rabbit hole as me :)