TheThingsNetwork/lorawan-stack #5818: Wrong LNS uri returned

RasmusGodske, 2022-09-22:
### Summary

**Background/Setup:**
I'm running lorawan-stack behind a reverse proxy in a kubernetes environment. The reverse proxy handles TLS, and receives traffic on the port 443, and unencrypts the requests and passes them to the non-tls ports(The ones starting with 18). So in short the stack, does not handle TLS at all. 

**Problem:**
However when querying the LNS connection endpoint `wss://example.com/router-info`, the reverse proxy processes and then redirects the request to the lorawan-stack at port 1887(non-TLS). The lorawan-stack responds with a wrong URI.

This is when passing this following configuration:
```
gcs.basic-station.default.lns-uri=wss://example.com:443
```

### Steps to Reproduce

**Launching and setting up gateway:**
1. Setup a reverse proxy which routes external traffic hitting port 443 to the lorawan-stack at port 1887
2. Launch the lorawan-stack with  `gcs.basic-station.default.lns-uri=wss://example.com:443`
3. Add new gateway in lorawan console `/console/gateways/my-new-gateway`
4. Add API key
5. Save API key

**Connecting to BasicStation - LNS Protocol:**
1. Using a websocket-client connect to `wss://example.com:443/router-info`
   1.1. Use following header: `{"Authorization": "{YOUR_API_KEY"}
2. Send following message: `{"router": "{YOUR_GATEWAY_EUI}"}`
3. Listen for message containing the URI from the websocket-server

### Current Result

The message containing the URI looks like this:

```json
{
   "router": "####:###:####:####",
   "muxs": "muxs-::0",
   "uri": "ws://example.com:80/traffic/eui-###############"
}
```

Notice how both the it is using `ws://` instead of `wss://` and port `80` instead of `443`

### Expected Result

I expected the message containing the URI to look like this:

```json
{
   "router": "####:###:####:####",
   "muxs": "muxs-::0",
   "uri": "wss://example.com:443/traffic/eui-###############"
}
```

### Relevant Logs

```shell
# Reverse Proxy Internal IP: 10.244.0.46

DEBUG   Run database query      {"duration": 0.001, "grpc.method": "GetIdentifiersForEUI", "grpc.service": "ttn.lorawan.v3.GatewayRegistry", "namespace": "db", "query": "SELECT id, created_at, updated_at, deleted_at, gateway_id, gateway_eui FROM \"gateways\"  WHERE \"gateways\".\"deleted_at\" IS NULL AND ((gateway_eui = $1)) ORDER BY \"gateways\".\"id\" ASC LIMIT 1", "request_id": "01GDJR61ZWZTSN8F5RH33JBVKT", "rows": 1, "source": "gateway_store.go:154", "values": ["##############"]}
INFO    Finished unary call     {"duration": 0.0019, "grpc.method": "GetIdentifiersForEUI", "grpc.service": "ttn.lorawan.v3.GatewayRegistry", "namespace": "grpc", "peer.address": "pipe", "request_id": "01GDJR61ZWZTSN8F5RH33JBVKT"}
DEBUG   Sent connection info response message   {"endpoint": "/router-info", "http.method": "GET", "http.path": "/router-info", "namespace": "web", "peer.address": "10.244.0.46:42748", "peer.real_ip": "10.244.0.46", "remote_addr": "10.244.0.46:42748", "request_id": "01GDJR61YSF4SCSRTDWASCZ37M"}
INFO    Request handled {"duration": 0.0371, "http.method": "GET", "http.path": "/router-info", "http.status": 200, "namespace": "web", "peer.address": "10.244.0.46:42748", "peer.real_ip": "10.244.0.46", "request_id": "01GDJR61YSF4SCSRTDWASCZ37M"}
```

### Deployment

The Things Stack Open Source (self-hosted)

### The Things Stack Version

3.21.2

### Client Name and Version

```shell
Using the CLI: 3.21.2
Using the docker image: thethingsnetwork/lorawan-stack:3.21.2
Using the reverse proxy: Traefik
Using the python websocket client: https://pypi.org/project/websocket-client
```

### Other Information

My suspision is that it gets the port and protocol, from the request it is receiving, instead of using the port defined in the: `gcs.basic-station.default.lns-uri`. 

I did look through the code at [discover.go#L61](https://github.com/TheThingsNetwork/lorawan-stack/blob/6cd5ac3ea012c39f7f89c6dfbd168d6f1ef83b85/pkg/gatewayserver/io/ws/lbslns/discover.go#L61), however I'm no GO expert and did not come to any conclusion.

### Contributing

- [X] I can help by doing more research.
- [ ] I can help by implementing a fix after the proposal above is approved.
- [X] I can help by testing the fix before it's released.

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

KrishnaIyer, 2022-09-23:
We have a config option for this. Please set `gs.basic-station.use-traffic-tls-address: true`. That will tell the Gateway Server to use TLS scheme in the traffic URL.

closed 2022-09-23 by KrishnaIyer

hm-dc, 2025-04-28:
I have the same issue with a reverse proxy using nginx. I have tried using the config option above and now the provided uri is wss://...:443. I need to redirect to ws://....1887

hm-dc, 2025-04-28:
I have fixed this using the `gs.basic-station.use-traffic-tls-address: true`  to force the gateway to connect to :443, and then a redirect in nginx to forward https://my.server:443/traffic/ on to port 1887 on the stack.