TheThingsNetwork/lorawan-stack #5844: BasicStation duty cycle computation mismatch

adriansmares, 2022-10-03:
### Summary

References https://github.com/lorabasics/basicstation/issues/164

BasicStation computes the sub band utilization in a different way compared to our sub band scheduler. This means that downlinks that we consider to be legal (i.e. within the duty cycle usage requirement), can be silently rejected by BasicStation.

[Production builds of BasicStation cannot be instructed to disable duty cycle checks](https://github.com/lorabasics/basicstation/blob/ba4f85d80a438a5c2b659e568cd2d0f0de08e5a7/src-linux/sys_linux.c#L755-L764). This means that we cannot just tell BasicStation to 'trust the LNS'.

### Current Situation

Downlink scheduling is a multi component affair. Fundamentally it goes between NS, GS and finally the packet forwarder. The flow is as follows:
1. NS tries to schedule a downlink on the GS.
2. GS accepts or rejects it. If rejected, go to 1 and try different gateway.
3. GS forwards it to the packet forwarder. If ack/nack, signal this to the LNS. This is just passed to the application layer, no retries are done (https://github.com/TheThingsNetwork/lorawan-stack/issues/18).

There are nuances to this. BasicStation does not actually have a nack message - only acknowledgements, and the ack is sent at the actual transmission time (so 5 seconds after the uplink for RX1=5), so even in the absence of the acknowledgement we cannot actually try to schedule again (since the window is closed).

Since BasicStation is non-cooperative in this regard, we need the GS to be 'as sure as possible' that if it schedules a downlink, the forwarder will accept it too. Otherwise the GS accepts it, the NS thinks that things are done, and then the forwarder silently drops it.

One way packet forwarder scheduling can fail is duty cycle. As described in the summary, there are actual differences between how we compute duty cycle and how BasicStation does it.

[BasicStation does not have rolling windows at all](https://github.com/lorabasics/basicstation/issues/164). BasicStation computes air time * 100/duty-cycle and considers that the next moment in time after which transmissions can take place. Any downlink between the current time and that point in the future is discarded silently. 

Consider the following scenario.
1. The current time is 00:00:00 (hh:mm:ss).
2. We want to schedule a downlink whose air time is 0.2 seconds, in a band with 1% duty cycle.

We compute 0.2 seconds * 100/1 = 20 seconds. This means that the next downlink in that particular sub band can only be transmitted after 00:00:20. Our scheduler is a bit more lenient - it will allow downlinks in this 20 seconds period, if the overall utilization in the rolling hour is under 1%.

### Why do we need this? Who uses it, and when?

Environments in which confirmed downlinks are used are affected by this issue the most. Due to the unfortunately symmetrical nature of this setup (every uplink airtime generates some downlink airtime), it is annoyingly easy to hit the duty cycle of a sub band.

In such situations, the GS will schedule (accept) the downlink, but the forwarder will silently drop it. This is very bad because generally there is overlapping coverage between multiple gateways, so if the GS would actually reject the scheduling attempt, the NS has the chance to try another gateway (which may not be duty cycled).

Confirmed downlinks are not the only ones that will hit this. Having duty cycle windows actually allows short bursts to occur - these are actually common ! Think of the initial moments of a session - you have a downlink for the join accept, then one or two downlinks for extra MAC commands (or even application layer setup). These are usually done in a short time frame, shorter than the Basic Station style allows.

### Proposed Implementation

Even if BasicStation would receive a bucket update, and we would align bucket sizes, most of the gateways in the field will not be upgraded for long periods of time. I think we need to accept this as a reality, and design and implement this in a pragmatic manner.

1. As part of `io.Frontend`, expose a `DutyCycleStyle() DutyCycleStyle` function that tells the GS which sub band scheduler to use. Valid styles are `Agnostic` (UDP, MQTT and gRPC are like this, since they don't do their own duty cycling) and `Blocking` (the BasicStation style mentioned above).
2. Abstract away the sub band scheduler under an interface, and turn the current `scheduling.SubBand` into a `scheduling.bucketSubBand`.
3. Introduce `scheduling.blockingSubBand` that follows the BasicStation rules. Statistics such as utilization are still based on hourly buckets, but scheduling attempts are computed in this 'sub band blocking' style.

### Contributing

- [X] I can help by doing more research.
- [X] I can help by implementing the feature after the proposal above is approved.
- [X] I can help by testing the feature before it's released.

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

adriansmares, 2022-10-03:
Pinging @johanstokking and @KrishnaIyer for thoughts on this, or if you see alternatives for the implementation. Otherwise thumbs up this message and we'll implement it as proposed.

beitler, 2022-10-06:
Hi @adriansmares , do you see major advantages of the windowed approach compared to the greedy approach implemented in LBS?

adriansmares, 2022-10-06:
> Hi @adriansmares , do you see major advantages of the windowed approach compared to the greedy approach implemented in LBS?

The greedy approach is not ideal for periodic steady states - i.e. saw tooth patterns. The greedy approach is more in line with straight lines associated with de-correlated request-response patterns. 

In reality we see lots of devices that do not decouple data collection from the data transmission and end up with periodic steady states. Of course that we don't send downlinks to each of those, but since the send windows are correlated, so become the receive windows (and even we respond to only 1% of devices, the pattern looks the same).

Now you may argue that maybe we shouldn't encourage these patterns and I fully agree, but I don't know if making duty cycle management more painful on a singular gateway is falling under 'nudge' - I think it is more of a bad experience.

adriansmares, 2022-10-06:
> Are your windows aligned with absolute time or the gateway-local xtick?

They are XTick based indeed. Given an emission at XTick `t`,  and a time on air `toa`, we check the usage in the `[t + toa - window, t + toa]` and `[t, t + window]` intervals.

PR #5862 (2022-10-12): BasicStation dutycycle alignment fixes

#### Summary

Closes #5844 

#### Changes

- Introduce the concept of dutycycle styles for the Gateway Server sub band scheduler.
  - We have two styles - window (what we have been using so far) and blocking window.
  - The blocking window style adds an additional check that ensures that BasicStation-based forwarders do not reject downlinks, since in BasicStation duty cycle is enforced by 'blocking' the sub band for the inverse duration of the duty cycle. What this 'inverse' means in plain words is that a 1 second transmission keeps a 1% duty cycle sub b [... cut]