TheThingsNetwork/lorawan-stack #5855: Error « outdated_data » on retrieving messages

allfree35, 2022-10-05:
### Summary

I use a LoRaWAN gateway from Kerlink which has the internal database backup functionality for Lora messages received when the wired Ethernet network is cut.
When reconnecting the gateway to the network, I only retrieve a small number of messages on my application while the gateway returns all the stored messages.
Most messages are "DROP" "Oudated_data" without knowing why some messages get through and not others.
It is not a chronological question since I recover messages after others are refused. I fetch the messages randomly, for example, during a test, I fetched the first message, another generated in the middle of the test and the last message.
I repeated the test several times, the result is always random.

### Steps to Reproduce

1) Use a LoRaWAN device to send every minutes a payload
2) Disconnect the gateway of the network for a short period (Test)
3) Reconnect the gateway to the network
4) See what happens on test messages

### Current Result

Most messages are "DROP" "Oudated_data" without knowing why some messages get through and not others.
It is not a chronological question since I recover messages after others are refused. I fetch the messages randomly, for example, during a test, I fetched the first message, another generated in the middle of the test and the last message.
I repeated the test several times, the result is always random.

### Expected Result

All messages retrieves should be accepted

### Relevant Logs

```shell
stack_1            | 2022-10-05T13:57:20.590673511Z WARN        Failed to update device in registry     {"ack": false, "adr": true, "adr_ack_req": false, "band_id": "EU_863_870", "bandwidth": 125000, "class_b": false, "data_rate": "lora:<bandwidth:125000 spreading_factor:8 > ", "dev_addr": "ALoNrA==", "device_channel_index": 4, "error": "error:pkg/networkserver:outdated_data (data is outdated)", "f_cnt_reset": false, "f_opts_len": 0, "f_port": 2, "frequency": 867300000, "full_f_cnt_up": 11, "grpc.method": "HandleUplink", "grpc.service": "ttn.lorawan.v3.GsNs", "last_f_cnt_up": 1, "m_type": "UNCONFIRMED_UP", "mac_count": 0, "mac_version": "MAC_V1_0_3", "major": "LORAWAN_R1", "namespace": "networkserver", "pending_session": false, "phy_payload_len": 24, "received_at": "2022-10-05T13:57:20.329948201Z", "request_id": "01GEM6ZGT93YFG07BERC4XD43D", "spreading_factor": 8, "uplink_f_cnt": 11}
stack_1            | 2022-10-05T13:57:20.591317121Z WARN        Finished unary call     {"duration": 0.2613, "error": "error:pkg/networkserver:outdated_data (data is outdated)", "error_correlation_id": "4149345df72b4f12b662264fdf4915c0", "error_name": "outdated_data", "error_namespace": "pkg/networkserver", "grpc.method": "HandleUplink", "grpc.service": "ttn.lorawan.v3.GsNs", "grpc_code": "FailedPrecondition", "namespace": "grpc", "peer.address": "pipe", "request_id": "01GEM6ZGT93YFG07BERC4XD43D"}
stack_1            | 2022-10-05T13:57:20.592762094Z WARN        Failed to update device in registry     {"ack": false, "adr": true, "adr_ack_req": false, "band_id": "EU_863_870", "bandwidth": 125000, "class_b": false, "data_rate": "lora:<bandwidth:125000 spreading_factor:8 > ", "dev_addr": "ALoNrA==", "device_channel_index": 1, "error": "error:pkg/networkserver:outdated_data (data is outdated)", "f_cnt_reset": false, "f_opts_len": 0, "f_port": 2, "frequency": 868300000, "full_f_cnt_up": 12, "grpc.method": "HandleUplink", "grpc.service": "ttn.lorawan.v3.GsNs", "last_f_cnt_up": 1, "m_type": "UNCONFIRMED_UP", "mac_count": 0, "mac_version": "MAC_V1_0_3", "major": "LORAWAN_R1", "namespace": "networkserver", "pending_session": false, "phy_payload_len": 24, "received_at": "2022-10-05T13:57:20.33048447Z", "request_id": "01GEM6ZGTADPYQ2J4D64BS2SDV", "spreading_factor": 8, "uplink_f_cnt": 12}
stack_1            | 2022-10-05T13:57:20.593249606Z WARN        Finished unary call     {"duration": 0.2627, "error": "error:pkg/networkserver:outdated_data (data is outdated)", "error_correlation_id": "9c32f07809044f149c47e31d7d005c9a", "error_name": "outdated_data", "error_namespace": "pkg/networkserver", "grpc.method": "HandleUplink", "grpc.service": "ttn.lorawan.v3.GsNs", "grpc_code": "FailedPrecondition", "namespace": "grpc", "peer.address": "pipe", "request_id": "01GEM6ZGTADPYQ2J4D64BS2SDV"}
stack_1            | 2022-10-05T13:57:20.595172963Z WARN        Failed to update device in registry     {"ack": false, "adr": true, "adr_ack_req": false, "band_id": "EU_863_870", "bandwidth": 125000, "class_b": false, "data_rate": "lora:<bandwidth:125000 spreading_factor:8 > ", "dev_addr": "ALoNrA==", "device_channel_index": 1, "error": "error:pkg/networkserver:outdated_data (data is outdated)", "f_cnt_reset": false, "f_opts_len": 0, "f_port": 2, "frequency": 868300000, "full_f_cnt_up": 13, "grpc.method": "HandleUplink", "grpc.service": "ttn.lorawan.v3.GsNs", "last_f_cnt_up": 1, "m_type": "UNCONFIRMED_UP", "mac_count": 0, "mac_version": "MAC_V1_0_3", "major": "LORAWAN_R1", "namespace": "networkserver", "pending_session": false, "phy_payload_len": 24, "received_at": "2022-10-05T13:57:20.331775746Z", "request_id": "01GEM6ZGTAZ36KASXN9ZARTF3J", "spreading_factor": 8, "uplink_f_cnt": 13}
stack_1            | 2022-10-05T13:57:20.596050171Z WARN        Finished unary call     {"duration": 0.2653, "error": "error:pkg/networkserver:outdated_data (data is outdated)", "error_correlation_id": "ece545c8076d44f6962478ec91739635", "error_name": "outdated_data", "error_namespace": "pkg/networkserver", "grpc.method": "HandleUplink", "grpc.service": "ttn.lorawan.v3.GsNs", "grpc_code": "FailedPrecondition", "namespace": "grpc", "peer.address": "pipe", "request_id": "01GEM6ZGTAZ36KASXN9ZARTF3J"}
stack_1            | 2022-10-05T13:57:20.597128341Z WARN        Failed to update device in registry     {"ack": false, "adr": true, "adr_ack_req": false, "band_id": "EU_863_870", "bandwidth": 125000, "class_b": false, "data_rate": "lora:<bandwidth:125000 spreading_factor:8 > ", "dev_addr": "ALoNrA==", "device_channel_index": 3, "error": "error:pkg/networkserver:outdated_data (data is outdated)", "f_cnt_reset": false, "f_opts_len": 0, "f_port": 2, "frequency": 867100000, "full_f_cnt_up": 18, "grpc.method": "HandleUplink", "grpc.service": "ttn.lorawan.v3.GsNs", "last_f_cnt_up": 1, "m_type": "UNCONFIRMED_UP", "mac_count": 0, "mac_version": "MAC_V1_0_3", "major": "LORAWAN_R1", "namespace": "networkserver", "pending_session": false, "phy_payload_len": 24, "received_at": "2022-10-05T13:57:20.340695046Z", "request_id": "01GEM6ZGTMXGW3P6BYH7CJHZJC", "spreading_factor": 8, "uplink_f_cnt": 18}
stack_1            | 2022-10-05T13:57:20.597558983Z WARN        Finished unary call     {"duration": 0.2568, "error": "error:pkg/networkserver:outdated_data (data is outdated)", "error_correlation_id": "7660074c9064406f9803ca61ce5a9d19", "error_name": "outdated_data", "error_namespace": "pkg/networkserver", "grpc.method": "HandleUplink", "grpc.service": "ttn.lorawan.v3.GsNs", "grpc_code": "FailedPrecondition", "namespace": "grpc", "peer.address": "pipe", "request_id": "01GEM6ZGTMXGW3P6BYH7CJHZJC"}
stack_1            | 2022-10-05T13:57:20.599261841Z WARN        Failed to update device in registry     {"ack": false, "adr": true, "adr_ack_req": false, "band_id": "EU_863_870", "bandwidth": 125000, "class_b": false, "data_rate": "lora:<bandwidth:125000 spreading_factor:8 > ", "dev_addr": "ALoNrA==", "device_channel_index": 3, "error": "error:pkg/networkserver:outdated_data (data is outdated)", "f_cnt_reset": false, "f_opts_len": 0, "f_port": 2, "frequency": 867100000, "full_f_cnt_up": 19, "grpc.method": "HandleUplink", "grpc.service": "ttn.lorawan.v3.GsNs", "last_f_cnt_up": 1, "m_type": "UNCONFIRMED_UP", "mac_count": 0, "mac_version": "MAC_V1_0_3", "major": "LORAWAN_R1", "namespace": "networkserver", "pending_session": false, "phy_payload_len": 24, "received_at": "2022-10-05T13:57:20.340936963Z", "request_id": "01GEM6ZGTM8CRJB71KWMATCJDR", "spreading_factor": 8, "uplink_f_cnt": 19}
stack_1            | 2022-10-05T13:57:20.599699874Z WARN        Finished unary call     {"duration": 0.2587, "error": "error:pkg/networkserver:outdated_data (data is outdated)", "error_correlation_id": "c6dfd6c9ee5a48c4bbc40af86bc42234", "error_name": "outdated_data", "error_namespace": "pkg/networkserver", "grpc.method": "HandleUplink", "grpc.service": "ttn.lorawan.v3.GsNs", "grpc_code": "FailedPrecondition", "namespace": "grpc", "peer.address": "pipe", "request_id": "01GEM6ZGTM8CRJB71KWMATCJDR"}
```

### Deployment

[... cut ...]
### The Things Stack Version

3.21.2

### Client Name and Version

```shell
Gateway: Kerlink IStation
Client: Web Browser Firefox 102.3.0esr (32 bits)
```

### Contributing

- [X] I can help by doing more research.
- [X] I can help by implementing a fix after the proposal above is approved.
- [X] I can help by testing the fix before it's released.

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

KrishnaIyer, 2022-10-11:
LoRaWAN uplink messages (and Join Requests with the Join Nonce) have Frame counters to prevent older messages from being processed, this is defined in the spec along with the reasons why.

So if you have a message that has a lower uplink frame counter than the highest seen value, that message is outdated by definition and will be dropped.

So how is your gateway flushing messages to TTS when it reconnects? Is it sending frame by frame in the order that it sees it or just dumping all messages?

I can see the situation where the gateway just dumps all the stored frames and there is no control of which frame is actually received by TTS first. 
In such a case, a frame with a higher Frame counter is seen and processed by TTS and all lower frame counts relative to that frame are dropped. 
Since the transmission of the frames on the backhaul is not controlled, a different random frame is received every time and the NS drops all frames with a relative lower frame count, which accounts for the randomness.

Unless you have some logs to show that TTS is dropping frames of a **higher than currently seen** frame counter value, this works as intended.

PS: LoRaWAN does not support buffering messages on gateways and then flushing them to the network.

allfree35, 2022-10-11:
Hi Krishna and thank you for your explanation. I better understand now the process and the imprtance of the frame counter.

The problem is that the gateway receive uplink messages in order as you can see :
Gateway Live Data
[image]

But the uplink messages are not treated in TTS application in the same order:
The first message treated is with F_cnt=12
Device Live Data
[image]

The next one: with F_cnt=8
[image]
The next one: with F_cnt=5
[image]

etc...

And finally the first message approuved by the application is the F_cnt=12 so all messages with F_cnt <12 are dropped.
Device Live Data
[image]

What i Think is, if all messages would be treated in a FIFO (First In First Out) process, I will be able to recover all messages.
Am I wrong in that suggestion?

Thank you.
Allan

KrishnaIyer, 2022-10-11:
I believe this is because the GS-NS communication is not synchronous and we use routines to handle traffic so we can't guarantee which routine completes first.

Can you share the Gateway Server log lines?

allfree35, 2022-10-11:
Yes I could give you my Gateway Server log lines, but I don't know where to have it.
I use self-hosted docker TTS open source. Usually I use this command to see logs:
 # docker-compose logs -t -f  stack
Could you give me the right way to the Gateway Server log ?

Allan

allfree35, 2022-10-12:
Hi,
there is nothing in the log except the disconnection of the gateway.
I added this section:
 "log.level: ' debug' 
in the stack global options in my docker-compose.xml but logs looks like the same that before.

allfree35, 2022-10-12:
Ok, I've found where to enable full DEBUG mode.
I do a new test and send you the logs

allfree35, 2022-10-12:
Debug is on
Nothing more:
$ grep "pkg/gatewayserver" error-outdated

^[[36mstack_1            |^[[0m 2022-10-12T13:35:28.600192280Z WARN     Task failed     {"error": "error:pkg/gatewayserver/io/udp:connection_expired (connection expired)", "gateway_eui": "7076FF0056070AD9", "gateway_ip_address": "100.100.165.103", "gateway_uid": "ren-1036-loragw3", "invocation": 6, "namespace": "gatewayserver/io/udp", "protocol": "udp", "remote_addr": "100.100.165.103:40804", "task_id": "disconnect_on_change_ren-1036-loragw3"}
^[[36mstack_1            |^[[0m 2022-10-12T15:22:14.438748422Z WARN     Task failed     {"error": "error:pkg/gatewayserver/io/udp:connection_expired (connection expired)", "gateway_eui": "7076FF0056070AD9", "gateway_ip_address": "100.100.165.103", "gateway_uid": "ren-1036-loragw3", "invocation": 1, "namespace": "gatewayserver/io/udp", "protocol": "udp", "remote_addr": "100.100.165.103:54136", "task_id": "disconnect_on_change_ren-1036-loragw3"}

KrishnaIyer, 2022-10-14:
Ok can you look for `"protocol": "ws"` if you're using a LoRa Basics Station gateway or `"protocol": "udp"` if you're using a UDP gateway?

allfree35, 2022-10-14:
I use UDP gateway.

This is the log of my last test in same condition. Only four/Ten messages DROP this time.

stack_1            | 2022-10-12T15:14:57.301054348Z INFO        Connected       {"gateway_eui": "7076FF0056070AD9", "gateway_ip_address": "127.127.165.103", "gateway_uid": "ren-1036-loragw3", "namespace": "gatewayserver/io/udp", "protocol": "udp", "remote_addr": "127.127.165.103:54136"}
stack_1            | 2022-10-12T15:15:03.587519696Z INFO        Downlink path claimed   {"gateway_eui": "7076FF0056070AD9", "gateway_ip_address": "127.127.165.103", "gateway_uid": "ren-1036-loragw3", "namespace": "gatewayserver/io/udp", "protocol": "udp", "remote_addr": "127.127.165.103:54136"}
stack_1            | 2022-10-12T15:16:41.049159334Z DEBUG       Synchronized server and gateway absolute time   {"concentrator_time": 1933700212000, "gateway_eui": "7076FF0056070AD9", "gateway_ip_address": "127.127.165.103", "gateway_time": 1665587801.0019999, "gateway_uid": "ren-1036-loragw3", "namespace": "gatewayserver/io/udp", "protocol": "udp", "remote_addr": "127.127.165.103:54136", "server_time": 1665587801.048727, "timestamp": 1933700212}
stack_1            | 2022-10-12T15:16:42.872815154Z DEBUG       Attempt to schedule downlink on gateway {"class": "CLASS_A", "gateway_eui": "7076FF0056070AD9", "gateway_ip_address": "127.127.165.103", "gateway_uid": "ren-1036-loragw3", "namespace": "gatewayserver/io/udp", "protocol": "udp", "remote_addr": "127.127.165.103:54136"}
stack_1            | 2022-10-12T15:16:42.872861300Z DEBUG       Attempt to schedule downlink in receive window  {"class": "CLASS_A", "data_rate": "lora:<bandwidth:125000 spreading_factor:7 > ", "frequency": 868300000, "gateway_eui": "7076FF0056070AD9", "gateway_ip_address": "127.127.165.103", "gateway_uid": "ren-1036-loragw3", "namespace": "gatewayserver/io/udp", "protocol": "udp", "remote_addr": "127.127.165.103:54136", "rx_window": 1}
stack_1            | 2022-10-12T15:16:42.872887330Z DEBUG       Computed scheduling delays      {"gateway_eui": "7076FF0056070AD9", "gateway_ip_address": "127.127.165.103", "gateway_uid": "ren-1036-loragw3", "median_rtt": null, "min_schedule_time": 0.13, "namespace": "gatewayserver/io/udp", "protocol": "udp", "remote_addr": "127.127.165.103:54136"}
stack_1            | 2022-10-12T15:16:42.872910219Z DEBUG       Computed downlink start timestamp       {"delay": 3.175864226, "gateway_eui": "7076FF0056070AD9", "gateway_ip_address": "127.127.165.103", "gateway_uid": "ren-1036-loragw3", "namespace": "gatewayserver/io/udp", "now": 1935524347774, "protocol": "udp", "remote_addr": "127.127.165.103:54136", "starts": 1938700212000}
stack_1            | 2022-10-12T15:16:42.872963192Z DEBUG       Scheduled downlink      {"class": "CLASS_A", "data_rate": "lora:<bandwidth:125000 spreading_factor:7 > ", "delay": 3.175864226, "duration": 0.071936, "frequency": 868300000, "gateway_eui": "7076FF0056070AD9", "gateway_ip_address": "127.127.165.103", "gateway_uid": "ren-1036-loragw3", "namespace": "gatewayserver/io/udp", "now": 1935524347774, "protocol": "udp", "remote_addr": "127.127.165.103:54136", "rx_window": 1, "starts": 1938700212000}
stack_1            | 2022-10-12T15:16:42.873074295Z DEBUG       Wait to schedule downlink message late  {"duration": 2.37590926, "gateway_eui": "7076FF0056070AD9", "gateway_ip_address": "127.127.165.103", "gateway_uid": "ren-1036-loragw3", "namespace": "gatewayserver/io/udp", "protocol": "udp", "remote_addr": "127.127.165.103:40804"}
stack_1            | 2022-10-12T15:16:45.249940681Z DEBUG       Write downlink message  {"gateway_eui": "7076FF0056070AD9", "gateway_ip_address": "127.127.165.103", "gateway_uid": "ren-1036-loragw3", "namespace": "gatewayserver/io/udp", "protocol": "udp", "remote_addr": "127.127.165.103:40804"}
stack_1            | 2022-10-12T15:16:45.990108944Z DEBUG       Received Tx acknowledgment, JIT queue supported {"gateway_eui": "7076FF0056070AD9", "gateway_ip_address": "127.127.165.103", "gateway_uid": "ren-1036-loragw3", "namespace": "gatewayserver/io/udp", "protocol": "udp", "remote_addr": "127.127.165.103:54136"}
stack_1            | 2022-10-12T15:16:49.307802434Z DEBUG       Synchronized server and gateway absolute time   {"concentrator_time": 1941963476000, "gateway_eui": "7076FF0056070AD9", "gateway_ip_address": "127.127.165.103", "gateway_time": 1665587809.2649999, "gateway_uid": "ren-1036-loragw3", "namespace": "gatewayserver/io/udp", "protocol": "udp", "remote_addr": "127.127.165.103:54136", "server_time": 1665587809.307339, "timestamp": 1941963476}
stack_1            | 2022-10-12T15:16:51.191771441Z DEBUG       Attempt to schedule downlink on gateway {"class": "CLASS_A", "gateway_eui": "7076FF0056070AD9", "gateway_ip_address": "127.127.165.103", "gateway_uid": "ren-1036-loragw3", "namespace": "gatewayserver/io/udp", "protocol": "udp", "remote_addr": "127.127.165.103:54136"}
stack_1            | 2022-10-12T15:16:51.192135730Z DEBUG       Attempt to schedule downlink in receive window  {"class": "CLASS_A", "data_rate": "lora:<bandwidth:125000 spreading_factor:7 > ", "frequency": 868500000, "gateway_eui": "7076FF0056070AD9", "gateway_ip_address": "127.127.165.103", "gateway_uid": "ren-1036-loragw3", "namespace": "gatewayserver/io/udp", "protocol": "udp", "remote_addr": "127.127.165.103:54136", "rx_window": 1}
[... cut ...] p", "protocol": "udp", "remote_addr": "127.127.165.103:40804"}
stack_1            | 2022-10-14T12:57:06.378225441Z DEBUG       Received Tx acknowledgment, JIT queue supported {"gateway_eui": "7076FF0056070AD9", "gateway_ip_address": "127.127.165.103", "gateway_uid": "ren-1036-loragw3", "namespace": "gatewayserver/io/udp", "protocol": "udp", "remote_addr": "127.127.165.103:54136"}

I noticed this "Failed to handle uplink message {"error": "error:pkg/gatewayserver/io:buffer_full (buffer is full)",

KrishnaIyer, 2022-10-14:
> I noticed this "Failed to handle uplink message {"error": "error:pkg/gatewayserver/io:buffer_full (buffer is full)"

So this indeed is the place where we throttle the GS - NS connections. This buffer handles 16 parallel messages per gateway and we don't offer an option to configure this since in LoRaWAN, gateways are not supposed to be sending that much traffic.

You can update this [here](https://github.com/TheThingsNetwork/lorawan-stack/blob/v3.22/pkg/gatewayserver/io/io.go#L39) in your local stack. Maybe set it to `1 << 6` (basically quadruple the buffers).

allfree35, 2022-10-14:
I thank you for your proposal and i would like to test it but I don't know where I can modify this in my configuration.
Self-Hosted docker-compose TTS.
Do I have to modify in the source and build from sources the docker container?

KrishnaIyer, 2022-10-14:
> Do I have to modify in the source and build from sources the docker container?

Yes indeed that's the method.

KrishnaIyer, 2025-04-30:
This issue is stale.