TheThingsNetwork/lorawan-stack #5860: Pasting in `Register end device` fields

ysmilda, 2022-10-10:
### Summary

When pasting in the fields on the `Register end device` page it concatenates the pasted values behind the existing values. While the behaviour I would expect is that it starts pasting/overwriting from the cursors position. 

With the current behaviour you'll have to clear the field first before pasting in another value.

### Steps to Reproduce

1. Fill a field on the `Register end device` page with some value.
2. Try pasting over it.

### Current Result

The value is concatenated.

### Expected Result

The value should be overwritten.

### Deployment

The Things Stack Community Edition

### The Things Stack Version

3.22.0

### Contributing

- [X] I can help by doing more research.
- [ ] I can help by implementing a fix after the proposal above is approved.
- [X] I can help by testing the fix before it's released.

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

ysmilda, 2022-10-10:
Additionally it is possible to paste in more than the maximum allowed characters which causes the character limit check to go off.

[image]

kschiffer, 2022-10-19:
@ysmilda Please try again. This should have been resolved via #5825. Otherwise please reopen.

ysmilda, 2022-10-24:
@kschiffer issues still exist in `v3.22.1`

kschiffer, 2022-10-25:
Note that bugs always go into the next version milestone.

closed 2022-11-08 by adriansmares