TheThingsNetwork/lorawan-stack #5861: Add configuration snippets for popular editors

ysmilda, 2022-10-10:
### Summary

To make life easier on starting developers it would be nice to share the editor settings that we have accumulated over time. This will ensure that others have access to the same checks as we do. This can either be done in the `DEVELOPMENT.md` or as complete configuration files. 

### Current Situation

N/A

### Why do we need this? Who uses it, and when?

This would be a great help for new developers as well as a way of documenting the settings used within the team.

### Contributing

- [X] I can help by doing more research.
- [X] I can help by implementing the feature after the proposal above is approved.
- [X] I can help by testing the feature before it's released.

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

nicholaspcr, 2022-11-09:
A few things that come to mind would be:
- A `editorconfig` file with all the style rules regarding all the languages that we use.
- A brefile with most, if not all, dependencies that we use for the day to day work.

We can have some sort of repository dedicated for both and refer to that in the `Development.md`

ysmilda, 2023-02-28:
For use of the `vscode-proto3` plugin in `vscode` the following needs to be added to the `settings.json` to have it work with the stack:

```json
"protoc": {
  "options": ["--proto_path=${workspaceRoot}/api"]
},
```