TheThingsNetwork/lorawan-stack #5875: Unable to log into console

mitya12342, 2022-10-16:
### Summary

I am trying to set up opensource edition of self-hosted things stack. I am using self-signed certificate with own CA and public IP address (no domain name). After following the [tutorial](https://www.thethingsindustries.com/docs/getting-started/installation/), I am able to log into admin account and /oauth interface, but not into the console, I am getting `Login failed` message. 

### Steps to Reproduce

1. Follow the open-source part of [tutorial](https://www.thethingsindustries.com/docs/getting-started/installation/)
2. Log in

I have used example open-source config files.
I am using IP address, so I replaced all `thethings.example.com` with server ip, placed correctly renamed ca, cert and key, commented and uncommented mentioned tls config sections, generated cookie keys. I can provide address of my installation if needed.

According to [Troubleshooting Installation Token Exchange Refused section](https://www.thethingsindustries.com/docs/getting-started/installation/troubleshooting/#token-exchange-refused):

1. I double checked `client-secret` (I have tried keeping it `console` and generating with `openssl rand -hex 32` )
2. I am not using localhost, but [this](https://www.thethingsindustries.com/docs/getting-started/installation/configuration/#solution-1-use-the-ip-address-of-your-computer-on-your-local-network) section hints that it should work on IP address (should it?).
3. Certificate is successfully validated against custom CA with `openssl verify -CAfile ca.pem cert.pem`; BUT it is only true for my own test CA/certs and fresh CA/certs generated by openssl. Method with cfssl from [docs](https://www.thethingsindustries.com/docs/getting-started/installation/certificates/#custom-certificate-authority) produces CA/certificates giving error 
```
error 18 at 0 depth lookup: self-signed certificate
error cert.pem: verification failed
```
4-5. I am running with own CA. 

I have tried to recreate everything with latest (3.22.0) and previous (3.21.2) version of docker container. Behaviours are the same.

### Current Result

When logging in, I get `Login failed` message with following details:
```
{
  "code": 7,
  "message": "error:pkg/web/oauthclient:exchange (token exchange refused)",
  "details": [
    {
      "@type": "type.googleapis.com/ttn.lorawan.v3.ErrorDetails",
      "namespace": "pkg/web/oauthclient",
      "name": "exchange",
      "message_format": "token exchange refused",
      "correlation_id": "bf85e97422e94d53950a895528dd21fb",
      "cause": {
        "namespace": "pkg/oauth",
        "name": "unauthorized_client",
        "message_format": "client is not authorized to request a token using this method",
        "correlation_id": "29d32f849283445baad6851a236dd635",
        "code": 7
      },
      "code": 7
    }
  ]
} 
```
And following messages in server log
```
stack_1     | WARN      OAuth internal error: get_client=client check failed, client_id=console {"http.method": "POST", "http.path": "/oauth/token", "namespace": "web", "oauth_client_id": "console", "oauth_error": "unauthorized_client", "peer.address": "172.19.0.1:58894", "request_id": "01GFGE6AJYX4EH4DETEWRKJ6JJ"}
stack_1     | WARN      OAuth error     {"error": "error:pkg/oauth:unauthorized_client (client is not authorized to request a token using this method)", "http.method": "POST", "http.path": "/oauth/token", "namespace": "web", "oauth_client_id": "console", "peer.address": "172.19.0.1:58894", "request_id": "01GFGE6AJYX4EH4DETEWRKJ6JJ"}
```
I can provide full logs with debug verbosity if needed.
Are there any way to display more details about oauth arrors?

### Expected Result

Successful console login

### URL

/console

### Deployment

The Things Stack Open Source (self-hosted)

### The Things Stack Version

3.22.0

### Client Name and Version

```shell
Using a web browser: Firefox 105.0.3 with own CA imported
```

### Contributing

- [X] I can help by doing more research.
- [ ] I can help by implementing a fix after the proposal above is approved.
- [X] I can help by testing the fix before it's released.

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

KrishnaIyer, 2022-10-18:
@mitya12342: Have you registered the console as an oauth client in the database?

mitya12342, 2022-10-18:
@KrishnaIyer Sure
Just to clarify
I did run
```
SERVER_ADDRESS=https://x.x.x.x
ID=console
NAME=Console
CLIENT_SECRET=xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
REDIRECT_URI=${SERVER_ADDRESS}/console/oauth/callback
REDIRECT_PATH=/console/oauth/callback
LOGOUT_REDIRECT_URI=${SERVER_ADDRESS}/console
LOGOUT_REDIRECT_PATH=/console
```
where x.x.x.x is my server ip
and xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx is random but the same as i put into `console.oauth.client-secret` of `ttn-lw-stack-docker.yml`
I also tried keeping it `console`
then after running
```
sudo docker-compose run --rm stack is-db create-oauth-client \
  --id ${ID} \
  --name "${NAME}" \
  --owner admin \
  --secret "${CONSOLE_SECRET}" \
  --redirect-uri "${REDIRECT_URI}" \
  --redirect-uri "${REDIRECT_PATH}" \
  --logout-redirect-uri "${LOGOUT_REDIRECT_URI}" \
  --logout-redirect-uri "${LOGOUT_REDIRECT_PATH}"
```
I got
```
Creating network "thingsstack_default" with the default driver
Creating thingsstack_redis_1    ... done
Creating thingsstack_postgres_1 ... done
Creating thingsstack_stack_run  ... done
INFO    Connecting to Identity Server database...
INFO    Updating OAuth client...
INFO    Updated OAuth client    {"secret": "aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa"}
INFO    Setting owner rights...
INFO    Set owner rights
DEBUG   No new version available
```
Do I need to set given secret `aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa` somewhere?

Thanks for your help!

KrishnaIyer, 2022-10-19:
> where x.x.x.x is my server ip
and xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx is random but the same as i put into console.oauth.client-secret of ttn-lw-stack-docker.yml

The client secret that you register for the console (` --secret "${CONSOLE_SECRET}" `) should be the same as the value used in the `--console.oauth.client-secret`.

Have you done that?

mitya12342, 2022-10-20:
Got it working

There is a typo in https://www.thethingsindustries.com/docs/getting-started/installation/running-the-stack/#initialization
I haven't noticed it until i typed everything by hand without copying it with variables
It sets CLIENT_SECRET and then passes CONSOLE_SECRET