TheThingsNetwork/lorawan-stack #5880: Device "Reset session and MAC state" on web console to clear DevNonce

descartes, 2022-10-19:
### Summary

The "Reset session and MAC state" button clears the session but not the DevNonce so a device that has been factory reset or a developer device needs jump-starting - either on the device or via the CLI.

It would be useful to have the web console clear the DevNonce as well

### Why do we need this? Who uses it, and when?

Not all users are CLI capable / happy and mirroring the CLI & web console functionality keeps things consistent.

### Contributing

- [ ] I can help by doing more research.
- [ ] I can help by implementing the feature after the proposal above is approved.
- [X] I can help by testing the feature before it's released.

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

johanstokking, 2022-10-26:
@descartes indeed, this button is in the Network Server section as it affects the current session only. This is also only relevant for ABP devices. What you are asking for is a DevNonce reset, which is only relevant for OTAA devices, and this would be part of the Join Server settings.

We already support this indeed via API and CLI, by setting `resets_join_nonces` (flag `--resets-join-nonces`).

Would it be helpful if we had a checkbox in the Console to configure this?

descartes, 2022-10-26:
> What you are asking for is a DevNonce reset, which is only relevant for OTAA devices, and this would be part of the Join Server settings.

Ah, yes, OK, this is why you get a seat on the committee table - a nuance to remember.

Although if the OTAA session is cleared which forces a join - should the button for OTAA be in the Join section and a clear counters button for ABP be in Network Server?

I already use --resets-join-nonces as Microchips MLS is rabid. @adriansmares tells me that a total reset option is coming soon to the CLI. But a button on the console to match would help those who don't CLI.

Many of these are "asking for a friend". But maybe I should apply to BBC Mastermind with a specialist subject on LoRaWAN 1.0.4 to force me to bone up further.

adriansmares, 2022-10-26:
> Would it be helpful if we had a checkbox in the Console to configure this?

We actually do, but it is visible only for LoRaWAN 1.1 devices for some reason. We could relax that.
https://github.com/TheThingsNetwork/lorawan-stack/blob/8ca5f15158393ef984c999f5d847866244281352/pkg/webui/console/views/device-general-settings/join-server-form/index.js#L166-L174

Regarding the CLI changes: `end-device reset` will also manually reset the `DevNonce` since https://github.com/TheThingsNetwork/lorawan-stack/pull/5829 , which is part of the `3.22.1` release. 

Console wise I think we should show the `resets-join-nonces` box always, and make the AS session reset part of the current `reset session and MAC state` button. I think that for end devices that have a NVM, but for some reason it has been erased, it is better to encourage a single time reset rather than enabling `resets-join-nonces`.

descartes, 2022-10-26:
> I think that for end devices that have a NVM, but for some reason it has been erased, it is better to encourage a single time reset rather than enabling `resets-join-nonces`.

Agreed, bit pointless having security if people fall in to bypassing it all the time. And if you are developing the CLI shouldn't be a stretch. FWIW, this came up with the Lacuna LS300 which has a serial interface and NVM and in testing someone got in to a muddle after resetting region & DevNonce count without realising the consequences.

johanstokking, 2022-11-01:
So we have two things here:

1. Showing the `resets_join_nonces` field for all LoRaWAN versions
2. Add a button to reset `used_dev_nonces` to an empty array. I think we **should not** reset `next_join_nonce`, because the end device may remember (some) server side nonces (AppNonce/JoinNonce)

PR #5958 (2022-12-02): Reset devNonce

#### Summary

Closes #5880 

#### Changes

- Show the `resets_join_nonces` field for all LoRaWAN versions
- Add a button to reset `used_dev_nonces`

[image]
[image]

#### Testing

Manully

##### Regressions

Showing the `resets_join_nonces` field for all LoRaWAN versions, instead of only for LoRaWAN 1.1 devices.

#### Checklist

- [ ] Scope: The referenced issue is addressed, there are no unrelated changes.
- [ ] Compatibility: The changes are backwards compatible with existing API, storage, configuration and CLI, according to the compatibility commitments in `README.md` for th [... cut]
closed 2022-12-02 by ryaplots