TheThingsNetwork/lorawan-stack #5882: "Mac Payload" term shouldn't be sure to indicate a "Frame Payload"

SylvainMontagny, 2022-10-19:
### Summary

In TTNv3, in the End Device Live data,  the term "MAC Payload" is used for the “user payload” whereas it should be written "Frame Payload" in accordance to the specification.

The TTN documentation also says so (obviously) : 
https://www.thethingsnetwork.org/docs/lorawan/message-types/#:~:text=MAC%20payload%20of%20the%20data%20messages 11](https://www.thethingsnetwork.org/docs/lorawan/message-types/#:~:text=MAC%20payload%20of%20the%20data%20messages

### Current Situation

"MAC Payload" term used instead of "Frame Payload"

### Desired Situation

Change "MAC Payload" by "Frame Payload" in the device live data.

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

adriansmares, 2022-10-21:
Which event types are rendered like this ? Can you share a picture of the places where this is mislabeled ?

SylvainMontagny, 2022-10-24:
oh, sorry. I should have updated my TTS version before.
[image]

Last version has corrected it :
[image]

You can close this issue.

Thanks for the nice work you are doing on this open-source stack.

Sylvain

closed 2022-10-25 by NicolasMrad