TheThingsNetwork/lorawan-stack #5906: Option to download/export the device's data in an application from the console

ymgupta, 2022-11-02:
### Summary

A few The Things Stack users requested an option for downloading/exporting the device's data in the application in `.csv` format.  

### Current Situation

Using The Things Stack CLI, we can get all the device's data in an application with the fields containing `id`, `created_at`, and `updated_at`. In addition to the default fields, other required fields can be fetced by providing the CLI flags of the fields. Below is the CLI command for reference. 
```
$ ttn-lw-cli end-devices list [application-id] [flags]
```

### Why do we need this? Who uses it, and when?

- Consolidated information on the registered devices in The Things Stack for the internal records in an application.
- Also, to know the consolidated information on the different types of devices in their tenant.
- For the manufacturers to make conclusions about which devices are in stock, and which are deployed.

### Proposed Implementation

An option to download/export the all device's details in an application from the Console

### Contributing

- [ ] I can help by doing more research.
- [ ] I can help by implementing the feature after the proposal above is approved.
- [X] I can help by testing the feature before it's released.

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

jpmeijers, 2022-11-03:
I'm actually currently running into a use for this. We are deploying >100 devices on site. To know which devices are working and being received by the gateway, we monitor the last seen time on the console. For >100 devices its impractical to scan through the pages of devices on the end device page. If one could download a csv of all devices in an application, along with their last see time, one can much easier analyse the data in excel.

KrishnaIyer, 2024-10-22:
With https://github.com/TheThingsNetwork/lorawan-stack/pull/7228, we have implemented a filter to fetch devices by the last `updated_at` value. This should cover the purposes of this issue.

closed 2024-10-22 by KrishnaIyer