TheThingsNetwork/lorawan-stack #5909: ttn-lw-cli end-devices list not including flags

jpmeijers, 2022-11-03:
### Summary

As mentioned in https://github.com/TheThingsNetwork/lorawan-stack/issues/5906 one should be able to run the command `ttn-lw-cli end-devices list [application-id] [flags]` to get an export of all devices in an application. I have tried this with the recommended set of flags from https://www.thethingsindustries.com/docs/getting-started/migrating/migrating-from-ce-to-ch/establish-new-session/ and that does not work. All I get is an export that includes the device_id, application_id, dev_eui, join_eui and created_at and updated_at.

### Steps to Reproduce

1. `$ ttn-lw-cli end-devices list my_application --name --description --lorawan-version --lorawan-phy-version --frequency-plan-id --supports-join --root-keys --mac-settings --limit 1000 > devices-ttn.json`

### Current Result

```
[{
  "ids": {
    "device_id": "<redacted>",
    "application_ids": {
      "application_id": "<redacted>"
    },
    "dev_eui": "<redacted>",
    "join_eui": "<redacted>"
  },
  "created_at": "2022-06-29T10:01:56.377Z",
  "updated_at": "2022-11-03T02:32:17.385920Z"
},
```

### Expected Result

The json should include:
* name
* description
* lorawan version
* phy version
* frequency plan
* otaa/abp
* mac settings

### Deployment

The Things Stack Community Edition

### The Things Stack Version

3.22.1

### Client Name and Version

```shell
The Things Network Command-line Interface: ttn-lw-cli
Version:             3.22.1
Build date:          2022-10-19T12:46:17Z
Git commit:          c24c6750a
Go version:          go1.19.1
OS/Arch:             linux/amd64
```

### Contributing

- [ ] I can help by doing more research.
- [ ] I can help by implementing a fix after the proposal above is approved.
- [X] I can help by testing the fix before it's released.

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

jpmeijers, 2022-11-03:
This issue is related to https://github.com/TheThingsNetwork/lorawan-stack-migrate/issues/52

These two issues both prevent me from batch migrating an application from TTS-CE to a TTS Cloud instance.

adriansmares, 2022-11-03:
`list` and `get` have different semantics due to the way the end device data is stored. There are no `List` / efficient projection for the NS/AS/JS entities, only IS entities, so a client `list` implementation like the CLI would have to manually do a `Get` to these stores. The migration guide uses only `get` for this exact reason, and without having `List` on the remaining stores, I am inclined not to fake it by doing potentially thousands of `Get` requests.

closed 2025-04-30 by KrishnaIyer