TheThingsNetwork/lorawan-stack #5932: Cannot paste Downlink Payload in console V3.22.2

echarlus, 2022-11-10:
### Summary

When using the web console, on the end device messaging / Downlink page it's not possible to paste ascii-encoded hexadecimal in the Payload Field. 
This used to work fine with previous versions ...

### Steps to Reproduce

In the webconsole select a provisionned and activated End device in an application
Go the Messaging and select Downlink
Select Push to Downlink queue
Enter a port value
In the Payload Type select Bytes 
Copy a valid hexadecimal text value and try to paste it into the Payload Text Field

### Current Result

The value is not pasted, field stays blank

### Expected Result

Value is pasted in the field and can then be submitted to device DL queue

### Deployment

The Things Stack Community Edition

### The Things Stack Version

v3.22.2 (be3a09ee0)

### Other Information

This used to work fine in previous versions 

### Contributing

- [X] I can help by doing more research.
- [ ] I can help by implementing a fix after the proposal above is approved.
- [ ] I can help by testing the fix before it's released.

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

adriansmares, 2022-11-11:
This should've been fixed and deployed yesterday based on https://github.com/TheThingsNetwork/lorawan-stack/pull/5928 . Could you please confirm / deny this ?

echarlus, 2022-11-11:
@adriansmares thanks, I've just checked and I can confirm it's now working as expected !

closed 2022-11-11 by adriansmares