TheThingsNetwork/lorawan-stack #5984: Fix failing activation of camera stream in QR code scanner in Console

kschiffer, 2022-12-20:
### Summary

The camera screen sometimes fails to load for reasons that are currently unknown

### Steps to Reproduce

https://youtu.be/zdSIBA8BehY?t=153

In the above video, one can see the camera feed failing to load. Although this is not universally reproducible and has to be investigated further.

### Current Result

Camera streams failing

### Expected Result

Camera stream activating as expected.

### Deployment

The Things Stack Community Edition

### The Things Stack Version

3.23.0

### Proposed Fix

This needs more research to find out reproduction steps.

### Contributing

- [X] I can help by doing more research.
- [X] I can help by implementing a fix after the proposal above is approved.
- [X] I can help by testing the fix before it's released.

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

adriansmares, 2023-05-23:
To be investigated and check if a fix/information is still needed in `v3.26.2`.

johanstokking, 2023-07-19:
Needs reproduction steps