TheThingsNetwork/lorawan-stack #6027: Developer Setup: Error in make init

oholthau, 2023-01-20:
### Summary

make init reponds with this error message:

go: go.thethings.network/lorawan-stack/v3@v3.0.0-00010101000000-000000000000 requires
        github.com/emersion/go-smtp@v0.16.0: missing go.sum entry; to add it:
        go mod download github.com/emersion/go-smtp
make: *** [tools/mage/mage.make:19: tools/bin/mage] Error 1

### Steps to Reproduce

1. Open https://github.com/TheThingsNetwork/lorawan-stack/blob/v3.24/DEVELOPMENT.md
2. Follow the instructions, set up development environment, clone repository
3. make init does the above error message

### Current Result

make init stops

### Expected Result

make init should run and finish

### Deployment

The Things Stack Open Source (self-hosted)

### The Things Stack Version

GitHub Branch 3.2.4 

### Contributing

- [X] I can help by doing more research.
- [ ] I can help by implementing a fix after the proposal above is approved.
- [ ] I can help by testing the fix before it's released.

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

adriansmares, 2023-01-26:
Hi @oholthau - what Go version do you have installed ? Could you please post the output of `go version` and `sudo go version` ?

The Things Stack requires Go version 1.18 at least.

oholthau, 2023-01-31:
Hi Adrian-Stefan,

I am using Ubuntu Server 22.04.

I had
go version go1.16.5 gccgo (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0 linux/amd64
but, in the dpkg list, it was go 1.18….

The solution was to install the package golang-go….

Thank you for your hint.

In the moment, I am fighting with Nodejs and npm…

BR
Olaf.

closed 2023-05-04 by adriansmares