TheThingsNetwork/lorawan-stack #6081: Enable telemetry after the entire telemetry collection flow is done

nicholaspcr, 2023-02-28:
### Summary

Currently the implementation of the telemetry exporter, collector and visualization is in progress. This issue has a purpose to serve as a reminder to enable the exporter on the Stack after everything is finished.

### Current Situation

Blocked by the other telemetry issues

### Desired Situation

To be enabled after the other issues are finished.

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

nicholaspcr, 2023-04-04:
Moved to the next milestone as https://github.com/TheThingsIndustries/lambda/issues/609 is still in progress and is needed for enabling the telemetry

nicholaspcr, 2023-06-29:
Moving this to the next milestone even with https://github.com/TheThingsIndustries/lambda/pull/962 merged, the main reason for this is to give time to deploy it to staging and eventually to production.

After the deployment to production the person who did the procedure will have the necessary information (tf outputs) to do the config PR in the Stack. Most likely that person will be @adriansmares so I'm putting you as an assignee on this issue

KrishnaIyer, 2023-07-12:
@nicholaspcr: This issue should focus on the following.

- [ ] The CLI commands seem to be slowed down when telemetry collection is enabled. So, similar to how we handle version updates, make these commands more performant.
- [ ] Add a clear Changelog with detailed information on this feature. Check with @adriansmares for the text.
- [ ] Enable the flag in OS.

KrishnaIyer, 2023-08-16:
@nicholaspcr: Since the enabling Telemetry is merged, this issue can be closed right?

nicholaspcr, 2023-08-16:
Indeed this is the case, closing the issue then.

closed 2023-08-16 by nicholaspcr