TheThingsNetwork/lorawan-stack #6093: Cannot login to TTN Console with self-signed certificates from docker compose

vAnArhist, 2023-03-07:
### Summary

Target platform:
1. Ubuntu 20.04
2. Docker Compose version v2.15.1
3. TTN (opensource) 3.24.1

### Steps to Reproduce

1. [Setup TTN server as localhost](https://www.thethingsindustries.com/docs/getting-started/installation/configuration/#running-the-things-stack-as-localhost) (with my custom static IP (**192.168.0.39**), setupped in ttn-lw-stack-docker.yml). 
2. TTN server installed as in [video](https://www.youtube.com/watch?v=Owm5IUtQTx8&t=269s&ab_channel=TheThingsNetwork), IP changed to my working PC-IP address.

### Current Result

After login to https://192.168.0.39 I have such result (see screenshot):
[image]
```
docker compose up
[+] Running 4/0
 ⠿ Network servertest_default       Created                                                                                0.0s
 ⠿ Container servertest-postgres-1  Created                                                                                0.0s
 ⠿ Container servertest-redis-1     Created                                                                                0.0s
 ⠿ Container servertest-stack-1     Created                                                                                0.0s
Attaching to servertest-postgres-1, servertest-redis-1, servertest-stack-1
servertest-redis-1     | 1:C 07 Mar 2023 11:39:42.973 # oO0OoO0OoO0Oo Redis is starting oO0OoO0OoO0Oo
servertest-redis-1     | 1:C 07 Mar 2023 11:39:42.973 # Redis version=7.0.9, bits=64, commit=00000000, modified=0, pid=1, just started
servertest-redis-1     | 1:C 07 Mar 2023 11:39:42.973 # Configuration loaded
servertest-redis-1     | 1:M 07 Mar 2023 11:39:42.974 * monotonic clock: POSIX clock_gettime
servertest-redis-1     | 1:M 07 Mar 2023 11:39:42.974 * Running mode=standalone, port=6379.
servertest-redis-1     | 1:M 07 Mar 2023 11:39:42.974 # Server initialized
servertest-redis-1     | 1:M 07 Mar 2023 11:39:42.974 # WARNING Memory overcommit must be enabled! Without it, a background save or replication may fail under low memory condition. Being disabled, it can can also cause failures without low memory condition, see https://github.com/jemalloc/jemalloc/issues/1328. To fix this issue add 'vm.overcommit_memory = 1' to /etc/sysctl.conf and then reboot or run the command 'sysctl vm.overcommit_memory=1' for this to take effect.
servertest-redis-1     | 1:M 07 Mar 2023 11:39:42.975 * Reading RDB base file on AOF loading...
servertest-redis-1     | 1:M 07 Mar 2023 11:39:42.975 * Loading RDB produced by version 7.0.9
servertest-redis-1     | 1:M 07 Mar 2023 11:39:42.975 * RDB age 6724 seconds
servertest-redis-1     | 1:M 07 Mar 2023 11:39:42.975 * RDB memory usage when created 0.82 Mb
servertest-redis-1     | 1:M 07 Mar 2023 11:39:42.975 * RDB is base AOF
servertest-redis-1     | 1:M 07 Mar 2023 11:39:42.975 * Done loading RDB, keys loaded: 0, keys expired: 0.
servertest-redis-1     | 1:M 07 Mar 2023 11:39:42.975 * DB loaded from base file appendonly.aof.1.base.rdb: 0.000 seconds
servertest-redis-1     | 1:M 07 Mar 2023 11:39:42.975 * DB loaded from incr file appendonly.aof.1.incr.aof: 0.000 seconds
servertest-redis-1     | 1:M 07 Mar 2023 11:39:42.975 * DB loaded from append only file: 0.000 seconds
servertest-redis-1     | 1:M 07 Mar 2023 11:39:42.975 * Opening AOF incr file appendonly.aof.1.incr.aof on server start
servertest-redis-1     | 1:M 07 Mar 2023 11:39:42.975 * Ready to accept connections
servertest-postgres-1  | 
servertest-postgres-1  | PostgreSQL Database directory appears to contain a database; Skipping initialization
servertest-postgres-1  | 
servertest-postgres-1  | 2023-03-07 11:39:43.104 UTC [1] LOG:  starting PostgreSQL 15.2 (Debian 15.2-1.pgdg110+1) on x86_64-pc-linux-gnu, compiled by gcc (Debian 10.2.1-6) 10.2.1 20210110, 64-bit
servertest-postgres-1  | 2023-03-07 11:39:43.104 UTC [1] LOG:  listening on IPv4 address "0.0.0.0", port 5432
servertest-postgres-1  | 2023-03-07 11:39:43.104 UTC [1] LOG:  listening on IPv6 address "::", port 5432
servertest-postgres-1  | 2023-03-07 11:39:43.105 UTC [1] LOG:  listening on Unix socket "/var/run/postgresql/.s.PGSQL.5432"
servertest-postgres-1  | 2023-03-07 11:39:43.109 UTC [30] LOG:  database system was shut down at 2023-03-07 11:23:39 UTC
servertest-postgres-1  | 2023-03-07 11:39:43.114 UTC [1] LOG:  database system is ready to accept connections
servertest-stack-1     | INFO   Setting up core component
servertest-stack-1     | WARN   No cookie hash key configured, generated a random one
servertest-stack-1     | WARN   No cookie block key configured, generated a random one
servertest-stack-1     | INFO   Setting up Identity Server
servertest-stack-1     | INFO   Setting up Gateway Server
servertest-stack-1     | INFO   Setting up Network Server
servertest-stack-1     | INFO   Setting up Application Server
servertest-stack-1     | INFO   Setting up Join Server
servertest-stack-1     | INFO   Setting up Console
servertest-stack-1     | INFO   Setting up Gateway Configuration Server
servertest-stack-1     | INFO   Setting up Device Template Converter
servertest-stack-1     | INFO   Setting up QR Code Generator
servertest-stack-1     | INFO   Setting up Packet Broker Agent
servertest-stack-1     | INFO   Setting up Device Repository
servertest-stack-1     | INFO   Setting up Device Claiming Server
servertest-stack-1     | INFO   Starting...
servertest-stack-1     | WARN   No cluster key configured, generated a random one       {"key": "c5c39a1cc48cf561a0ba59a983d01427a6270de4f1c09c5f806bc188a2f8d1b9", "namespace": "cluster"}
servertest-stack-1     | INFO   Listening for connections       {"address": ":1884", "namespace": "grpc", "protocol": "gRPC"}
servertest-stack-1     | INFO   Listening for connections       {"address": ":8884", "namespace": "grpc", "protocol": "gRPC/tls"}
[... 42 lines cut ...]
```shell
Using both Firefox v110.0.1 and Chrome browser
ttn-lw-stack version
The Things Stack for LoRaWAN: ttn-lw-stack
Version:             3.24.1
Build date:          2023-02-16T09:29:24Z
Git commit:          bd83cb329
Go version:          go1.19.5
OS/Arch:             linux/amd64
```

### Other Information

Folder with docker compose configuration and keys:
[serverTest.zip](https://github.com/TheThingsNetwork/lorawan-stack/files/10908984/serverTest.zip)

### Contributing

- [ ] I can help by doing more research.
- [ ] I can help by implementing a fix after the proposal above is approved.
- [ ] I can help by testing the fix before it's released.

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

KrishnaIyer, 2023-03-08:
The issue seems that the Console is not able to fetch an Oauth token from the Oauth server. 
Can you double check if the secret used for the console is the same as the one that is registered while setting up the Identity Server? 
Basically check the params of the `ttn-lw-stack is-db create-oauth-client` command.

vAnArhist, 2023-03-08:
@KrishnaIyer After several attempts it started to work. Thanks for help!

closed 2023-03-08 by vAnArhist

mjlaurent6, 2023-04-02:
hi can you share more how you fix this ? im actually stuck on similar problems

the diff is i use go command to run since i want to try on dev server (not sure how to change code etc workflows using docker-compose)