TheThingsNetwork/lorawan-stack #6094: incorrect display of "downlink_count" on console and API

sjunger, 2023-03-07:
### Summary

Hello,
the "downlink_count" currently has a bug. The display in the console fluctuates between "3" and the actual continuous value.

### Steps to Reproduce

The effect can be seen in the "Gateway" console.
After a few reloads of the page, the value jumps back and forth.
This also happens in the query via API.

### Current Result

Screenshots on Slack #support.

### Expected Result

Of course, only the correct current value should be displayed here.

### URL

https://eu1.cloud.thethings.network/console/gateways/<Gateway-ID>

### Deployment

The Things Stack Community Edition

### The Things Stack Version

3.24.2

### Contributing

- [ ] I can help by doing more research.
- [ ] I can help by implementing a fix after the proposal above is approved.
- [ ] I can help by testing the fix before it's released.

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

KrishnaIyer, 2023-03-08:
This requires more information. Can you please post screenshots/API responses? 
Please hide information such as EUIs when doing so. Thanks.

sjunger, 2023-03-08:
here are the missing screenshots:
[image]
[image]

KrishnaIyer, 2023-03-09:
Ok thanks. @kschiffer can you take a look?

sjunger, 2023-03-09:
Hi,

I have no idea what's going on here.

Yesterday at 2pm the connection was reset and now the data is reliable again. The bug was "silently" fixed.

I will now inevitably process my previous statistics data again and import them into a new influxdb table.
I hereby close the ticket.

Greetings Stefan

[image]

closed 2023-03-09 by sjunger