TheThingsNetwork/lorawan-stack #6099: Save search results when accessing a device and going back

NicolasMrad, 2023-03-09:
### Summary

When users use the search in the console and press on an end-device from the search list, upon going back to the previous page the search result is cleared out. It would be more efficient to keep the search results live.

### Current Situation

When going to the previous page, which should be the search result, you go back to the end-device list

### Why do we need this? Who uses it, and when?

So users don't have to go and search for the same keyword again

### Contributing

- [ ] I can help by doing more research.
- [ ] I can help by implementing the feature after the proposal above is approved.
- [X] I can help by testing the feature before it's released.

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

adriansmares, 2023-05-23:
Can we save it just in the current Redux store ? I would say that the point is not that this is some kind of user data, but really per-browsing session (i.e. it is erased when you close the tab or refresh).

kschiffer, 2023-05-24:
It just needs to be saved/interpreted as query parameters. That way it is also possible to send other users a link to the results a specific page of a specific search query.