TheThingsNetwork/lorawan-stack #6111: Add `class_c_enabled` for class C mode

johanstokking, 2023-03-13:
### Summary

Add `class_c_enabled` to the current MAC state indicate whether class C is enabled for 1.0.x devices.

Add `class_c_always_enabled` to the end device (stored in NS). This should be taken from a new field in the Device Profile (part of TS002 1.2).

### Current Situation

Currently, NS uses the end device's `supports_class_c` flag for 1.0.x devices.

### Why do we need this? Who uses it, and when?

We need the `class_c_enabled` flag to indicate that class C is actually enabled right now. This is used for downlink scheduling.

We need the `class_c_always_enabled` flag to set the `class_c_enabled` flag to an initial value.

In 1.1 or higher, `DeviceModeInd` still takes precedence.

### Proposed Implementation

The flag `supports_class_c` should only be used to indicate whether the end device implements class C support. This is probably not relevant for the NS (for now). So what should be removed, is NS logic that uses `supports_class_c`.

Instead, the logic to check whether class C is enabled should be determined based on this new flag in the MAC state.

The `class_c_always_enabled` flag is simply added to the end device itself. It serves as a default for `class_c_enabled` in a new MAC state. Again, only for 1.0.x devices.

### Contributing

- [X] I can help by doing more research.
- [X] I can help by implementing the feature after the proposal above is approved.
- [X] I can help by testing the feature before it's released.

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

adriansmares, 2023-05-23:
> The `class_c_always_enabled` flag is simply added to the end device itself. It serves as a default for `class_c_enabled` in a new MAC state. Again, only for 1.0.x devices.

In the absence of `class_c_always_enabled`, the default value should still be `true` for backwards compatibility, right ?

Also, since there is no `DeviceModeInd` in 1.0.x, how would a device signal to the NS the switch ?

johanstokking, 2023-05-24:
> Also, since there is no `DeviceModeInd` in 1.0.x, how would a device signal to the NS the switch ?

The application does that, so that would go via the AS. So the AS would be able to set this field on the NS, autonomously based on payload, so without forwarded auth from a user request.