TheThingsNetwork/lorawan-stack #6112: Support `administrative_contact` and `technical_contact` in the Console

nicholaspcr, 2023-03-14:
### Summary

References #4930

With the addition of  Administrative/Technical in #4996, there is the need to add support for it on the Console. 

This means that `Clients`, `Gateways`, `Applications` and `Organizations` would benefit from the support of the `administrative_contact` and `technical_contact` fields and that `Users` have their contact information deprecated. Furthermore this implies in organization not sending emails to a set of users but to a specific email determined by the admin user.

With this in mind, there is the need to update the entities mentioned to support the `administrative_contact` and the `technical_contact`.

### Current Situation

Currently there are pages withing these entitites: `Application`, `Gateway`, `Organizaiton` which are attached to either a User with email or a organization with a set of Users.

### Why do we need this? Who uses it, and when?

Add the support for `administrative_contact` and the `technical_contact` within those entities, allowing to trigger emails to a specific target instead of the whole set of users that belong to that organization.

### Proposed Implementation

I lack the knowledge to know what is the best approach on the front-end, but I imagine that adding fields similar to the `email_address` the user's profile section but applied to administrative_contact and technical_contact would be appropriate.

### Contributing

- [ ] I can help by doing more research.
- [ ] I can help by implementing the feature after the proposal above is approved.
- [X] I can help by testing the feature before it's released.

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

adriansmares, 2023-04-04:
Blocked by https://github.com/TheThingsIndustries/lorawan-stack/issues/3623

ryaplots, 2023-04-12:
Are these fields related to any user? Or is this restricted to admins only?

adriansmares, 2023-04-12:
The fields are relevant to any collaborators with the `{ENTITY}_SETTINGS_BASIC` right - they can edit the said contact.

kschiffer, 2023-04-25:
I'm a bit confused that this was scheduled before the UX was ready or signed-off, this is what we have the `needs/ux` label for. Let's make sure that we follow the proper flow with new features. This probably lacked a bump for my attention to address the UX.

cc @nicholaspcr

NicolasMrad, 2023-04-26:
@kschiffer maybe I should have asked you about it, but it's because every time I asked about the progress of this issue it was communicated that it's going well without mentioning that it's missing UX from your side. As discussed this will be moved to the next milestone until the UX is done.

kschiffer, 2023-04-26:
Blocked by #5027

PR #6162 (2023-07-26): Support `administrative_contact` and `technical_contact` in the Console

#### Summary

Closes #6112 

#### Changes

- Add contacts to relevant entities.

<details>
<summary>Screenshots</summary>
[image]
[image]
</details>

#### Testing

Manually

#### Notes for Reviewers

I decided to add this in the entity general settings, as this is the place where I have access to the collaborator's rights for the entity, and not in the user settings as we initially thought.

#### Checklist
closed 2023-07-26 by ryaplots