TheThingsNetwork/lorawan-stack #6113: Make PBA use `administrative_contact` and `technical_contact` instead of old `contact_info`

nicholaspcr, 2023-03-14:
### Summary

References #4930 

With #4996 replacing 'Contact_info' with 'administrative_contact' and 'technical_contact' there is the need to update the `PacketBrokerAgent` to use the respective field.

### Current Situation

There is the usage of the deprecated `contact_field`.

### Desired Situation

The support for the two new fields  `administrative_contact` and `technical_contact` while disregarding the old field of `contact_info`.

There is the confusion introduced by https://github.com/TheThingsNetwork/lorawan-stack/issues/4930#issuecomment-1195312377 where I'm not sure what the TTSC is.  @johanstokking does this mean that that the `contact_info` will be dropped and replaced by an alternative form that registers the admin and technical email?  Or a form that determines the main email that should receive all the notifications?

@adriansmares Might know that TTSC is and the comment may be clear to you. If so, I would like to ask a bit of clarification on this.

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

adriansmares, 2023-03-14:
> There is the confusion introduced by [#4930 (comment)](https://github.com/TheThingsNetwork/lorawan-stack/issues/4930#issuecomment-1195312377) where I'm not sure what TTSC is. @johanstokking does this mean that that the `contact_info` will be dropped and replaced by an alternative form that registers the admin and technical email? Or a form that determines the main email that should receive all the notifications?
> 
> @adriansmares Might know that TTSC is and the comment may be clear to you. If so, I would like to ask a bit of clarification on this.

TTSC = The Things Stack Cloud, i.e. `(eu1|eu2|nam1|au1).cloud.thethings.industries`.
TTSCE = The Things Stack Community Edition, i.e. `(eu1|nam1|au1).cloud.thethings.network`.

The complication introduced by `administrative_contact`/`technical_contact` is that they are a layer of indirection - you no longer have access directly to names and emails. You need to actually retrieve the user / organization and find out what the email is. To make the matters even better, you can end up in a situation in which you need to follow up chains of administrative contains (entity -> organization -> organization -> ... -> organization -> user) until you reach an email. 

To make things even funkier, I am not sure we're disallowing loops.

We may need an RPC with cluster authentication which allows a recursive lookup of these details first. Otherwise it is very hard to actually use `administrative_contact`/`technical_contact`.

---

What @johanstokking was mention in that PR is that Packet Broker needs that email address at the moment, while PBA has no way of doing the user lookup. I guess that we can either go via the RPC route, or we remove the email check from PB somehow.

johanstokking, 2023-03-22:
Let's not bother with this for now; we might just remove this from Packet Broker indeed, as it doesn't have a real use today.

closed 2025-04-30 by KrishnaIyer