TheThingsNetwork/lorawan-stack #6120: missing RX metadata

NicolasMrad, 2023-03-22:
### Summary

missing RX metadata in the (non verbose) device traffic events

### Current Situation

Can’t see the RX metadata of join-requests, maybe due to the routing policy or duty cycle

### Why do we need this? Who uses it, and when?

To be able to see the RX metadata of join-requests

### Proposed Implementation

up for discussion when triaged

### Contributing

- [ ] I can help by doing more research.
- [ ] I can help by implementing the feature after the proposal above is approved.
- [ ] I can help by testing the feature before it's released.

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

johanstokking, 2023-03-28:
The issue is that when a device activates while looking at the application or device traffic view in the Console, a join-accept event is shown. This, for good reasons, does not include RX metadata of the join-request.

It would be very helpful to see RX metadata to figure out why a device didn't receive the join-accept. So what would be helpful, is showing join-request events.

From what I understood from an offline discussion, is that the verbose event stream does include the join-request message, at least on the device level and maybe also on the application level. I'll check this out. If that is indeed the case, we can probably just close this issue, or decide to show the join-request in non-verbose mode.

johanstokking, 2023-03-28:
I think that for debugging purposes, it is very helpful to see the join-request. It's the first thing the (OTAA) end device can do. I'd be in favor of including it in the default filter and propagating it to the parent.

PR #6320 (2023-06-13): Missing RX metadata
#### Summary
Extend the default end device and application filters in the Console with the `ns.up.join.process` events from the NS. Propagate the same events to application level.

Closes #6120 

#### Changes
- Added `ns.up.join.process` events to the Console filter
- Propagated the same events to parent

#### Testing
- Events are displayed in the end device and application Live Data without turning on verbose streaming and contain the `rx_metadata` field 

##### Regressions

...

#### Notes for Reviewers

...

#### Checklist

- [ ] Scope: The referenced issue is addressed, [... cut]