TheThingsNetwork/lorawan-stack #6128: Deprecate NaN and (-)Infinity usage in payload decoders

adriansmares, 2023-03-27:
### Summary

Payload decoders may emit `NaN` and `Infinity` or `Infinity` number values, as these are supported values in JavaScript. 

The most commonly used serialization format is JSON, and JSON cannot represent these values without creating a confusion between `NaN`, the float concept, and `NaN`, the string. 

We should consider whether or not allowing these values as part of the output is valuable or not.

### Current Situation

Historically we've rendered these values as `"NaN"` and `"Infinity"`, but our newer `protojson` library [does not](https://github.com/protocolbuffers/protobuf-go/commit/0fd4f3a506894c05a9cad2d52c521a8ce953b49d) like to render these values as strings anymore, as they break the round-trip concept (serialize -> deserialize should be the identity function).

The current workaround is to round-trip the values via `structpb.NewStruct` which [still](https://github.com/protocolbuffers/protobuf-go/blob/808c66411fe76c839a68168654228446fbdc1ecf/types/known/structpb/struct.pb.go#L401-L413) allows the special values to be rendered as strings. This is not ideal as it changes the protobuf serialization result, but it does maintain the JSON serialization.

Regarding the usage today:
1. `Infinity` is never observed as part of our stored messages.
2. `-Infinity` is virtually never observed (10^-9) as part of our stored messages.
3. `NaN` is observed in under 0.001% as part of our stored messages.

This methodology is biased towards the storage integrations, but historically we haven't seen any `NaN` or `Infinity` usage in the wild which has been intentional. 

Generally `NaN`, which is the most observed (but still barely observed overall) is a result of a mathematical operation which interacts with a list and goes out of bounds. Consider the following snippet, which can occur while consuming bytes from an end device too eagerly (i.e. consuming what is not there).
```js
a = [1, 2, 3] 
console.log(a[4] * 5) // This outputs NaN.
```

### Desired Situation

We should remove support for these values. We have at least two options here:
1. Sanitize recursively the result object.
- PRO: if only some parts of the decoded payload are wrong, we still output these values.
- CON: the remaining part of the message may still be garbage, since the logic may depend on evaluating such degenerate NaNs. It is also the case that we're no longer rendering fields which historically were there.
2. Drop the whole payload.
- PRO: downstream consumers do not see potentially invalid decoded payloads.
- CON: it could be the case that some parts of the decoded message are still useful.

I am more open to option 2, and if any other opinions are present, I am ready to hear them.

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

johanstokking, 2023-03-28:
My opinion is to unset the field if it's NaN or an inifinity and set a warning.

The intention is a numeric field. If the codec is using operators on non-numeric fields, you get NaN. If the codec is using operators with very small or very large values, likely erroneously, you get an infinity. In both cases, a string value is unexpected. We did have a work-around for conveying the JavaScript NaN/inifinity values in JSON, but it breaks JSON decoding in places where a numeric field is expected. That is most likely unintended; I don't see a valid reason to intentionally convey NaN and infinity from a codec to the application layer.

johanstokking, 2023-03-28:
> I am specifically afraid of libraries which are configured to read `"NaN"` as NaN and actually use this as part of their logic today. I don't know how they will handle missing values and I hope they won't end up interpreting `0` as being a valid value (since they didn't unmarshal NaN).

This is problem A. Problem B is that applications built in typed languages expect a numeric value but they get a string and so processing fails.

I find problem B worse than A.

NaN should really not be used to convey that a value isn't there. That's really what `undefined` is for.

In the Device Repository, `NaN` is only used in polyfills, input checks and other general airthmetic checks. That's all legitimate use. No codec test expects `NaN`.

adriansmares, 2023-03-28:
> NaN should really not be used to convey that a value isn't there. That's really what `undefined` is for.

`null`, or literary missing the object key, as `undefined` does not exist in JSON either. 

I do see your point. I think that with the current warnings (event warnings, Console warnings, changelog warnings) we should be able to phase them out. 

I do not see a good use for `NaN` that would convey a missing value, and I can't think of useful `Infinity` or `-Infinity` values either.

---

Another challenge - what if the `NaN` is part of a list - do we delete the list element, or do we put something else (`null` ?) instead ?

We once again risk breaking logic by simply deleting the element. I think that using a `null` is perhaps better, just to keep the structure intact.

johanstokking, 2023-03-28:
> > NaN should really not be used to convey that a value isn't there. That's really what `undefined` is for.
> 
> `null`, or literary missing the object key, as `undefined` does not exist in JSON either.

I meant to say that if a codec wants to return no message, or unintentionally by an arithmetic operation on `undefined` leading to `NaN`, it can return `undefined`, in which case the field won't be set in the JSON output.

> Another challenge - what if the `NaN` is part of a list - do we delete the list element, or do we put something else (`null` ?) instead ?
> 
> We once again risk breaking logic by simply deleting the element. I think that using a `null` is perhaps better, just to keep the structure intact.

Good point. This is something that makes me doubt if we should skip fields ...

We can set the element to `null` or unset the list field entirely. I'd be most inclined going with the former indeed.

But if we change element values NaN to `null`, shouldn't we do that with normal fields as well?

adriansmares, 2023-03-28:
> But if we change element values NaN to `null`, shouldn't we do that with normal fields as well?

For consistency reasons, we probably should indeed. The problem is that `null` probably can't be unmarshalled to numbers either (unless a nullable value was accepted in the first place).

johanstokking, 2023-03-29:
I think we need to decide between:

1. Discarding the decoded payload entirely and emit an error
2. Omitting all NaN fields or arrays with a NaN

I think indeed that sending `null` for a numeric field is almost as bad as sending a string.

adriansmares, 2023-03-29:
I think we should add some temporary Prometheus metrics in `v3.25.1` that would count how often this happens and in which situations.

At the moment we care about:
- Type of invalid value (`NaN`, `Infinity`, `-Infinity`).
- Context (object vs list)

Are there other dimensions we should be counting over ?

adriansmares, 2023-07-07:
We should keep this issue open in order to track the status of these values, based on metrics. We did not remove the support yet.

adriansmares, 2023-08-16:
I've looked at metrics using the following two expressions:
```
sum by (payload_formatter, value_context, value_type) (
  rate(ttn_lw_as_uplink_payload_value_violations_total[5m])
)
```

```
sum by (payload_formatter, value_context, value_type) (
  rate(ttn_lw_as_downlink_payload_value_violations_total[5m])
)
```

Key takeaways from the data of the last 7 days:
- Downlinks are virtually unaffected by this.
- Uplinks are affected by this have payload formatters coming from JavaScript, not the Device Repository. There are virtually no relevant instances of Device Repository formatters returning `NaN`s/`Infinity`s.
- As expected, in the only instances where this happens, it is generally the `NaN`s which appear in `structs`. There is an outlier case in TTSCE `nam1` where we actually have `Infinity`s, but otherwise it's only `NaN`s in `structs`.
- As implied above, there aren't actually relevant instances of lists containing `NaN`s/`Infinity`s.
- TTS Community Edition has more instances of this happening, probably because JS formatters are more DIY for deployments done there. TTSC is virtually unaffected.
- Overall, the rates of decoded payloads which contain these values are negligible compared to the traffic rates in the clusters themselves.

I would say this is almost a non issue at the end of the day, and in the future I would probably prefer to fail the decoding as a whole in order to force the remaining formatters to be fixed. Let me know your thoughts.

johanstokking, 2023-08-16:
I'd agree with failing decoding.

closed 2023-10-11 by adriansmares