TheThingsNetwork/lorawan-stack #6140: Using Node V18.13 gives error

LDannijs, 2023-04-04:
### Summary

Building the stack using Node V18.13.0 returns an error: `Error: error:0308010C:digital envelope routines::unsupported`

[image]

Using node V16.20 with nvm does work

### Steps to Reproduce

1. follow the steps here while downloading the newest Node version from their website: https://github.com/TheThingsNetwork/lorawan-stack/blob/v3.25/DEVELOPMENT.md#development-environment
2. run `tools/bin/mage js:build`

### Current Result

Returns `Error: error:0308010C:digital envelope routines::unsupported`

### Expected Result

Build correctly without fail

### Deployment

The Things Stack Open Source (self-hosted)

### The Things Stack Version

3.24.2

### Contributing

- [X] I can help by doing more research.
- [ ] I can help by implementing a fix after the proposal above is approved.
- [ ] I can help by testing the fix before it's released.

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

pablojimpas, 2023-04-14:
I've encountered this issue as well, using both with [`v16.20.0`](https://archlinux.org/packages/community/x86_64/nodejs-lts-gallium/) and [`v18.15.0`](https://archlinux.org/packages/community/x86_64/nodejs-lts-hydrogen/). I had to go all the way back to [`v14.21.3`](https://archlinux.org/packages/community/x86_64/nodejs-lts-fermium/).

Seems like at some point Node.js migrated to OpenSSL 3.0, essentially breaking the algorithm that Webpack 4 uses for hashing.

There are workarounds to use newer Node versions, the easiest is setting this environment variable:

```sh
export NODE_OPTIONS=--openssl-legacy-provider
```

The solution here would be migrating to Webpack 5 at some point, since it has support for the newer Node versions (and supposedly is [backwards compatible all the way back to Node v10.13.0](https://webpack.js.org/concepts/#environment))

kschiffer, 2023-04-17:
We currently support [up to 16.20 LTS](https://github.com/TheThingsNetwork/lorawan-stack/blob/v3.25/DEVELOPMENT.md#development-environment), as the developer documentation states. If just tried building the frontend from scratch using `tools/bin/mage js:clean js:cleanDeps jsSdk:clean jsSdk:cleanDeps js:build` and it ran through without issues.

Updating to webpack 5 is not an option at the moment as there are still (last time I checked) too many incompatibilities that would make this task blow up way too much. I will reconsider this in the future though.

Closing this for now as such.

closed 2023-04-17 by kschiffer