TheThingsNetwork/lorawan-stack #6150: RekeyConf not transmitted

cprovidenti, 2023-04-12:
### Summary

After transmitting a Join-Accept to a LoRaWAN 1.1.0 end-device, and receiving a (Confirmed) uplink from that end-device that contains the required RekeyInd MAC command, the Network Server transmits no (more) downlinks to that end-device. In particular, the Network Server never transmits the expected RekeyConf MAC command in response, so that after sending ADR_ACK_DELAY (= 64) uplinks, all of which have a RekeyInd MAC command, the end-device re-enters the Join state as called for by the LoRaWAN 1.1 specification. At that point, the preceding sequence of events repeats itself.

**Note:** As far as I can tell, this issue is the same as issue #5343. The symptoms, at least, are identical. As noted below (see Other Information), issue #5343 known to have been fixed subsequently, so perhaps the cause of this issue is different from that of issue #5343.

### Steps to Reproduce

1. Have a LoRaWAN 1.1 end-device join The Things Network using OTAA.
2. Wait until that end-device has sent ADR_ACK_DELAY (= 64) uplinks, and watch it re-enter the Join state.
3. Allow that end-device to join again.
4. While the end-device is sending those ADR_ACK_DELAY uplinks, (try to) send a downlink to that end-device.

**Note:** Issue #5343 outlines another (probably more convenient) way of reproducing the problem.

### Current Result

The following screenshot of the verbose (live data) stream for that end-device shows an occurrence of the problem with The Things Stack Version identified in this issue (as opposed to the earlier version identified in issue #5343):

[image]

### Expected Result

The Network Server transmits a RekeyConf MAC command in response to the RekeyInd MAC command from a LoRaWAN 1.1 end-device. I.e., same as for issue #5343.

### Deployment

The Things Stack Community Edition

### The Things Stack Version

3.25.1

### Other Information

The end-device uses LoRaMac-node.git v4.7.0.

An earlier (unrelated) test result in my possession shows that this issue was *not* present in *at least one* earlier version of The Things Stack Community Edition, namely: v3.21.1.

### Contributing

- [X] I can help by doing more research.
- [ ] I can help by implementing a fix after the proposal above is approved.
- [ ] I can help by testing the fix before it's released.

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

adriansmares, 2023-04-13:
Thanks for the report ! I can confirm that this is indeed a regression, and the temporary fix in the original comment ( https://github.com/TheThingsNetwork/lorawan-stack/issues/5343#issuecomment-1081877391 ) should fix it.

https://github.com/TheThingsNetwork/lorawan-stack/pull/6152 should ensure that we don't regress again on this issue, and it will fix it for newer sessions.

cprovidenti, 2023-04-13:
If you would like me to close the issue, just say so. As is, I assume that you will close the issue once the next milestone is released, which is fine by me.

adriansmares, 2023-04-13:
The issue will be closed once the PR is merged, and the fix will be available once the next version is released and deployed, but I would like to know if the original 'temporary fix' works for you.

cprovidenti, 2023-04-13:
The original 'temporary fix' (namely: to configure explicitly values for ADR_ACK_DELAY and ADR_ACK_LIMIT) works for me too. E.g., see screenshot:

[image]

closed 2023-04-14 by adriansmares