TheThingsNetwork/lorawan-stack #6165: Show conflicting identifiers information in the AS/NS/JS end device registries

adriansmares, 2023-04-20:
### Summary

The End Device Registries enforce a globally unique constraint on the `(join_eui, dev_eui)` pair for each individual end device. This means that even if someone registers the end device in with another device ID, in perhaps another application or another tenant, the `(join_eui, dev_eui)` unique constraint cannot be violated.

The AS/NS/JS end device registries do enforce this uniqueness constraint, but their error message is very raw, containing only the fact that a conflict exists, but not with whom.

The IS end device registry is a lot more friendly and on end device creation actually provides all of the conflicting information to the caller. We should do the same in the AS/NS/JS registries.

### Current Situation

The IS will provide the caller with the conflicting end device ID, application ID and even tenant ID if needed. See the following logic:

https://github.com/TheThingsIndustries/lorawan-stack/blob/84ad8dc3a911fa686c9eda00e45aa7107b7a9bc1/pkg/identityserver/end_device_registry.go#L218-L251

### Why do we need this? Who uses it, and when?

The AS/NS/JS do have access to this information:
https://github.com/TheThingsNetwork/lorawan-stack/blob/ac161143c2f27e9266416f5353b50b6d09bb71c5/pkg/networkserver/redis/registry.go#L782-L795

Right now an `EXISTS` command is used, but if we would use a `GET`, we would know the full end device UID, which can be parsed back to the identifiers using `unique.ToDeviceID` and `unique.ToTenantID`.

### Proposed Implementation

Instead of `EXISTS`, use `GET` and parse the end device UID using `unique.ToDeviceID` and `unique.ToTenantID`. Try to keep the returned errors similar to the IS format, and avoid code duplication. There is a `pkg/internal/registry` which can serve as a common root for the logic if needed.

### Contributing

- [X] I can help by doing more research.
- [ ] I can help by implementing the feature after the proposal above is approved.
- [X] I can help by testing the feature before it's released.

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

KrishnaIyer, 2023-05-09:
I'm currently working on the End Device Registries for a separate item and this is related. So I'll pick this up.

KrishnaIyer, 2023-05-09:
> Instead of EXISTS, use GET and parse the end device UID using unique.ToDeviceID and unique.ToTenantID. Try to keep the returned errors similar to the IS format, and avoid code duplication. There is a pkg/internal/registry which can serve as a common root for the logic if needed.

adriansmares, 2023-05-09:
> I think we should only return the device/app ID if it's registered by the same tenant.

I consider that we should always show the conflicting tenant ID. The information is not secret (EUIs are public, and they are visible over the air anyway), and in general we show the source of the conflict for all conflicts. The original PR (https://github.com/TheThingsIndustries/lorawan-stack/pull/2507) has the same spirit.

As a user I do not have any form of recourse if my end device is already registered, so showing who is causing the conflict is important for resolution.

PR #6359 (2023-07-14): Backport: Show conflicting identifiers in AS/NS/JS
#### Summary
Provide error information in the AS/NS/JS device registries on uniqueness constraint violation.

Closes #6165

#### Changes
- Added attributes to the error definition

#### Testing
- Existing UT

##### Regressions
...

#### Notes for Reviewers

...

#### Checklist

- [ ] Scope: The referenced issue is addressed, there are no unrelated changes.
- [ ] Compatibility: The changes are backwards compatible with existing API, storage, configuration and CLI, according to the compatibility commitments in `README.md` for the chosen target b [... cut]
closed 2023-07-14 by cvetkovski98