TheThingsNetwork/lorawan-stack #6180: WARN    Failed to check version update  {"error": "Get \"https://www.thethingsindustries.com/releases/thethingsstack-open-source.json\": dial tcp 18.65.202.102:443: i/o timeout"}

CropWatchDevelopment, 2023-04-28:
### Summary

While spinning up a new docker image (latest) I am getting the following error:

`WARN    Failed to check version update  {"error": "Get \"https://www.thethingsindustries.com/releases/thethingsstack-open-source.json\": dial tcp 18.65.202.102:443: i/o timeout"}`

The reason I am thinking that the issue isn't not with my firewall is because I am able to access: `https://www.thethingsindustries.com/releases/thethingsstack-open-source.json` via my browser, and I am unable to access:
`https://www.thethingsindustries.com/releases/thethingsstack-open-source.json\`. Perhaps the `\` got in there by accident?
Or did I possibly do someting wrong to accidentily add the rogue back slash?

### Steps to Reproduce

Newly installed Ubuntu Server OS
Install Docker
Copy/Past contents of: [https://www.thethingsindustries.com/docs/the-things-stack/host/docker/configuration/docker-compose-open-source.yml](https://www.thethingsindustries.com/docs/the-things-stack/host/docker/configuration/docker-compose-open-source.yml)
create directory structure and upload stack.
run `docker-compose pull`
run `docker-compose run --rm stack is-db migrate`
See issue

### Current Result

I see the message: `"error": "Get \"https://www.thethingsindustries.com/releases/thethingsstack-open-source.json\": dial tcp 18.65.202.102:443: i/o timeout"`

### Expected Result

No error

### Relevant Logs

```shell
docker-compose run --rm stack is-db migrate
/snap/docker/2746/lib/python3.6/site-packages/paramiko/transport.py:32: CryptographyDeprecationWarning: Python 3.6 is no longer supported by the Python core team. Therefore, support for it is deprecated in cryptography. The next release of cryptography (40.0) will be the last to support Python 3.6.
  from cryptography.hazmat.backends import default_backend
Creating thethingsstack_stack_run ... done
INFO    Connecting to Identity Server database...
WARN    Failed to check version update  {"error": "Get \"https://www.thethingsindustries.com/releases/thethingsstack-open-source.json\": dial tcp 18.65.202.102:443: i/o timeout"}
failed to connect to `host=postgres user=root database=ttn_lorawan`: dial error (timeout: dial tcp 172.18.0.2:5432: connect: connection timed out)
    correlation_id=3bdf43aceca44519991627ed3d6e86fd
ERROR: 255
```

### Deployment

The Things Stack Enterprise (self-hosted)

### Client Name and Version

```shell
The Things Stack for LoRaWAN: ttn-lw-stack
Version:             3.25.1
Build date:          2023-04-18T08:13:17Z
Git commit:          bcfbbc610
Go version:          go1.20.3
OS/Arch:             linux/amd64
```

### Other Information

Not sure if this error is on my end, or the containers, but from the error message, it seems there may be an ending back slash. Is this correct?

### Contributing

- [X] I can help by doing more research.
- [X] I can help by implementing a fix after the proposal above is approved.
- [X] I can help by testing the fix before it's released.

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

adriansmares, 2023-05-01:
The original error message is `Get "https://www.thethingsindustries.com/releases/thethingsstack-open-source.json": dial tcp 18.65.202.102:443: i/o timeout`, but because it has to be rendered as a JSON string, it needs to be wrapped between `"`. This in turn would make pairing the `"` not possible, so the original `"` becomes a `\"` in order to clarify where the string starts and where it ends. This is called escaping.

---

Back to the original warning - Is this a recurring issue ? Network errors are not completely uncommon.

CropWatchDevelopment, 2023-05-05:
I got things working after some changes to my docker config.I am going to say I was doing something wrong, and will close this.

closed 2023-05-05 by CropWatchDevelopment