TheThingsNetwork/lorawan-stack #6205: Add End Device pictures via the Console

KrishnaIyer, 2023-05-09:
### Summary

Add End Device pictures via the Console

### Current Situation

https://github.com/TheThingsNetwork/lorawan-stack/pull/3701 from a long time ago adds Console support for user-uploaded Profile pictures (And Gravatar).

We have similar API and backend support for user-uploaded picture for an End Device, presumably to keep track of where/how the device was installed. 

We currently support uploading end device pictures via the CLI and the API.

### Why do we need this? Who uses it, and when?

For users to actually use this feature. Currently, no one seems uses this since there is no Console support.

### Proposed Implementation

- Check Backend (IS) and API if something needs update.
- Implement console support.

### Contributing

- [ ] I can help by doing more research.
- [ ] I can help by implementing the feature after the proposal above is approved.
- [ ] I can help by testing the feature before it's released.

### Code of Conduct

- [X] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

johanstokking, 2023-05-09:
Just like QR code scanning, can we use the phone's camera to take a picture when using the Console on mobile?

kschiffer, 2023-08-31:
Currently working on this as part of my current bigger UX refactor.

KrishnaIyer, 2025-04-30:
Let's not bother with this for now

closed 2025-04-30 by KrishnaIyer